st-one-io/node-red-contrib-s7 #2: S7 node not show on Raspberry PI

kusumy, 2016-11-29:
I have Raspberry PI Model B, with node-red installed version 0.15.2.
And I have installed node-red-contrib-s7 using npm install. 
The installation was successful, but I can not see the S7 node on the Node-Red.

I've check the Node-Red start-up console, and here the message:

/ui
/usr/lib/node_modules/node-red-contrib-s7/red/s7.js:198
Object.keys(data).forEach(key => sendMsg(data[key], key, null));
^
29 Nov 08:11:11 - [warn] ------------------------------------------------------
29 Nov 08:11:11 - [warn] [s7] SyntaxError: Unexpected token >
29 Nov 08:11:11 - [warn] ------------------------------------------------------

Strangely, I tried to install on the Ubuntu 16.04, and I can see the S7 node on the Node-red.

gfcittolin, 2016-11-29:
Oh, that's probably because of the arrow function, part of the ES6 standard, available by default only on newer versions of Node.js. Could you please confirm which version of the Node executable do you have on raspberry?
I'll push a change to use a normal function here, but meanwhile you can try upgrading your Node.js there

gfcittolin, 2016-11-30:
@kusumy please update your installation with the new version 0.2.2. I've removed the arrow function, and it should now work with older versions of Node.JS. Let me know if you find any other problem

closed 2016-11-30 by gfcittolin