st-one-io/node-red-contrib-s7 #3: Error: read ECONNRESET

trinitry, 2017-03-07:
hello,
i am currently using/testing the S7 node which by the ways i found it very usefull and well done for this. Lately, i run into uncaught exception, more specifically 
i get the following somehow randomly, and if it keeps on 
> [error] [s7 endpoint:ITK_104] Failure (Bad values)
  i finally get

> [red] Uncaught Exception:
7 Mar 13:58:54 - Error: read ECONNRESET
    at exports._errnoException (util.js:870:11)
    at TCP.onread (net.js:544:26)

and node-red crashes..
do you have any ideas on how i could track and solve this?
running on
npm -v 4.3.0
Node-RED version: v0.16.2
Node.js  version: v4.2.4
Linux 4.4.13-yocto-standard ia32 LE
Dashboard version 2.1.0 started at /ui

thanks in advance, any help would me more than welcome.

gfcittolin, 2017-03-09:
Hello,

The TCP connection to the PLC is handled by the underlying [plcpeople/nodeS7](https://github.com/plcpeople/nodeS7) library. Therefore, we probably need to check the code there. Do you have the full stack trace of this exception?

But, aside of that, an ECONNRESET usually means that the other side (here the PLC) has closed the connection. Are you maybe reaching the limit of the connections to the PLC? Are you connecting to the CPU or to a CP? Which model?

trinitry, 2017-03-10:
Hello,
i am pretty sure the issue comes from the latest node-red v0.16.2 upgrade. What i did is that i clone a new mmc card with the node-red 0.15.3 and run it for a night without any issue. I upgrade the node-red to the latest and on the same overall system running the same code the problem appear. I did notice thought that during upgrade some errors about glyp appear that seemed the installer took care.
How i could produce a full stack trace?
I just have the output from the log of node-red.
It is connected to the CPU (S71200), i am thinking that the issue is not from the PLC side.
Thank you in advance

hbrianjr, 2017-04-18:
I am also experiencing the error message "Error connecting to PLC: Error: read ECONNRESET".  This has only started since the latest node-red v0.16.2.  As trinity noted above, node-red v0.15.3 and v0.14.x did not exibit this issue.  Any help is appreciated.

gfcittolin, 2017-04-19:
I'm currently on a business trip and I'll be back by the weekend. so can check this only next week. Meanwhile, could one of you provide me the flow you're using for me to simulate your configuration?

*Note*: Since v0.2.4 you can enable the debugging output of the underlying nodes7 library by passing -v when running node-red, or by setting `verbose: true` on settings.js. Would be also interesting to see the library's output when the error happens

icarogregorio, 2017-05-04:
Hi,

Updating the node-red-contrib-s7 then appears a timeout field that I set it to 4000 milleseconds.
When I do this the error ECON Reset stops.
A important note: When you are working with S7 node you have to use Deploy a modified flows.

gfcittolin, 2017-06-04:
I'm closing this issue, since the only possibility of throwing this has been fixed upstream. If any of you still has such problems after upgrading to the latest versions of `node-red-contrib-s7` (currently `1.1.2`) and `nodes7` (currently `0.1.15`), please let me know and I reopen it

closed 2017-06-04 by gfcittolin