st-one-io/node-red-contrib-s7 #5: Compatible with Logo 0ba8?

finn112, 2017-05-31:
Hi, 

is this node also compatible with the Siemens Logo 0ba8?
If so, what changes are necessary? How do I control virtual inputs of the logo?

Are there any instructions?

Many thanks for your help!

gfcittolin, 2017-06-04:
Hi,

Unfortunately, this is not possible right now. As far as I know, we'd need to be able to set the local and remote TSAP numbers to initiate a connection with Logo devices, but the library we use does not support it right now.

Actually, this is not complicated to implement, but we unfortunately have no Logo device so we could test with. If you have this possibility, Pull Requests are welcome!

finn112, 2017-06-04:
Hello gfcittolin,

I have a logo. I also like to help. What do I have to do exactly. How do I find the requests?

finn112, 2017-06-04:
Sorry, first use. :-D

gfcittolin, 2017-06-05:
Hi @finn112 

I've prepared some code for you to try out. Please do the following:
 - uninstall your current node-red-contrib-s7 package (or have a separate node-red installation for the tests)
 - install a separate branch by running `npm install github:netsmarttech/node-red-contrib-s7#tsap`

So now you might have a slightly different version of the node, allowing to select different "modes" for connection. Please then try using the TSAP mode for connecting to the Logo (should also work with S7-200 series). Remember that the node's Remote TSAP is the PLC's Local TSAP an vice versa. Maybe [these instructions](https://support.industry.siemens.com/cs/document/56348902/how-do-you-establish-a-connection-between-a-logo!-0ba7-and-an-hmi-panel-?dti=0&lc=en-DE) from Siemens may help

If you get a connection... cheers! :) You can try out reading data from the Logo then. I've never used a Logo directly, so I don't know exactly how these Virtual Inputs are mapped, but I remember reading somewhere that it maps to a DB1, so maybe you could try with addresses like `DB1,X0.0` and so on.

If something goes wrong, try running Node-RED in verbose mode by running `node-red -v`. This enables verbose mode of the underlying library, so we can try to see what went wrong. You can also see the needed changes by checking the differences between `master` and `tsap` branches in both [netsmarttech/node-red-contrib-s7](https://github.com/netsmarttech/node-red-contrib-s7/tree/tsap) and [netsmarttech/nodeS7](https://github.com/netsmarttech/nodeS7/tree/tsap).

Looking forward the results of your test! 👍

gfcittolin, 2017-06-05:
@plcpeople maybe you also want to check this out...

plcpeople, 2017-06-05:
This is interesting, I hadn't considered communication to these PLCs with the nodeS7 library.

I see some virtual memory address info at this address https://www.promotic.eu/en/pmdoc/Subsystems/Comm/PLC/LOGO.htm
But I have not verified any of this as we don't have one of these for testing either.  

Note that as it is now, nodeS7 uses a local TSAP of "0100" (looks like this should be 0200 for the Logo) and a remote TSAP that starts with 01 and the second part is the slot number, which defaults to 2 but this is usually specified as a connection parameter as it usually depends on the type of PLC.  

If the sample code works I will merge the TSAP code in to the nodeS7 library as this could be very useful for others.

finn112, 2017-06-06:
Hey gfcittolin, hi plcpeople,

Thanks for your help. I uninstalled the node. But i can't install the other version.
http://www.bilder-upload.eu/show.php?file=d0b95a-1496774022.jpg

plcpeople, 2017-06-06:
Try something like:
npm install git://github.com/netsmarttech/node-red-contrib-s7#tsap

finn112, 2017-06-06:
I had node-red running on a windows system. Have now changed to the Raspberri Pi 3. Tomorrow I'll try the communication.

gfcittolin, 2017-06-07:
It looks like you don't have git installed on your computer. As an
alternative, you can just download everything as a zip file from git,
unpack it in your node-red installation, and just run `npm install` from
inside it. Note that it must be inside your Node-RED installation, that is:
either in the node_modules folder where node-red is installed, or inside
the node_modules folder of the node-red's user directory
(%HOMEPATH%\.node-red in Windows)

finn112, 2017-06-08:
Hi guys, 

I have tried the following:
Logo and Node, as can be seen in the pictures.
This guide I have still found, since it is apparently the same. Unfortunately I could not see any change in the node.
http://www.frickelzeugs.de/logocontrol/konfiguration/

What comes next?

[image]
[image]
[image]
[image]
[image]
[image]

gfcittolin, 2017-06-09:
Hi,

Well, it looks like you already got a connection, and this is very good! Half the way alread!

Maybe you can try setting the addres of Q1 to `DB1,X1064.0`, this way it should map correctly to the VM area of the logo. But I have to say I'm not into the internals of Logo, so maybe there's something else.

Tip: You can also run Node-RED in verbose mode with `-v` flag or by setting `verbose: true` in `settings.js` of Node-RED. This turns nodes7's debugging on and we can see more details about the communication.

Also: Do you see any errors in the Debug tab of Node-RED, of then in the console, when running Node-RED?

finn112, 2017-06-09:
Hi,

it works! I just had to change the address. After the weekend, I will deal with the other direction.
Thanks a lot!

[image]
[image]

gfcittolin, 2017-06-09:
Excelent! I'm very happy it worked for you. I'll clean up some things and merge/PR the work in the master branches, as it should be backwards-compatible with installations that do not set the TSAP fields directly.

Please keep us informed about the other tests you're going to make :)

gfcittolin, 2017-06-09:
Hey guys,

The TSAP functionality just landed in `1.3.0`. Some minor fixes and other functionality (like import/export the variables) has also been merged. You should be able to use it with Logo by using a simple `npm install node-red-contrib-s7` or by Node-RED's interface.

Thank you by helping improving the node and by testing it!

closed 2017-06-09 by gfcittolin

ImeaHungary, 2017-08-12:
Hi guys,
I was try out the latest version with the Logo!8 and digital I/O can read perfectly, but it is strange that is work in any TSAP setting (now is not so big problem for me).
But I still can't read the analogue values..... I mean Analog input (AI 1032 ~ 1064) or VM...
It could be good to read the NAQ (Network Analogue Output) 1406~1469  64 bytes... this values can read from the - maybe just the setting was wrong...
Can somebody give me some experience?
The 
[image]

ImeaHungary, 2017-08-12:
The above picture shows that the analogue flags are now set in my Logo, but I can't reach it from Node-Red...

plcpeople, 2017-08-12:
What address are you using in Node-Red to try to get this working?  DB1,INT0 should get VM0.  DB1,INT1118 should get AM1.

ImeaHungary, 2017-08-12:
Dear plcpeople!
Thank's for your reply - you are right - it is working by your suggestion!

gfcittolin, 2017-08-14:
Thanks @plcpeople for helping out! 👍 

As soon I get some time, I'll try to write out a small readme with some instructions on how to correctly define the address for both the S7-300/400/1200 and for the Logo 8, so it may get easier for everybody when configuring the addresses.

Ideally we may also get a kind of address validation/translation at the time of configuring the S7 Endpoint on Node-RED, I just need to find the time for doing this :)

DuncanCT, 2017-11-08:
Is it possible to control network inputs? Or any inputs for that matter with Logo!8?

gfcittolin, 2017-11-08:
I don't think you can control (write to) network inputs, just read them. In any case, the address would be starting at `DB1,X1246`, according to the already mentioned https://www.promotic.eu/en/pmdoc/Subsystems/Comm/PLC/LOGO.htm

Let us know if you manage to do it :)

DuncanCT, 2017-11-08:
Sorry, I don't know enough about this stuff but I'm hacking around at the moment tyring to get it to work and the only thing I've noticed that the debug boolean injection into a NI is different from output debug of the NI. See below... 

[image]

Would it make a difference if the injection was structured differently?

DuncanCT, 2017-11-09:
You can write to digital inputs if you add stuff to Soft Comfort. I'm using it with an IoT2020. 

In Soft Comfort add the IoT as a new device>S7-compatible device>complete name and ip details.

In the diagram add the NI block, Edit Block Properties>Remote Device>Select IoT from device list>TSAP (I selected 03.00 but it doesn't seem to effect the outcome)>Block Type V>VB:0 Bit:0

Node > PLC Variables > DB1,X1246.0

VB:1 = DB1,X1246.1 etc...

MaartenV87, 2018-05-19:
Hi Duncan, did you manage to control the IoT2020 with nodered?
I have a Siemens Logo 8 and woulld like to control it with HomeAssistant. Any advice on a workaround?

DuncanCT, 2018-05-19:
Hey Maarten, In my example you can control the Logo from Node-RED via the IoT2020. I'm not sure about direct control from Node-RED to a Logo. Assume that's what you are asking?

IBernsdorf, 2018-06-25:
Good morning guys, 
I have similar issue. I am trying to connect node red to a Siemens LOGO 8.2 without success. It's not possible to read out flags or other information. Do you have an idea to solve this problem?

IBernsdorf, 2018-06-25:
I am using a Raspberry 1 model A and and Node Red 0.18.7

gfcittolin, 2018-06-27:
What kind of error you're having? Can you post your configuration here, so we can check it out?

rogelioRuiz, 2018-06-28:
Hi, I've been able to control the VM0 with node-red however the Network Input is acting as a momentary switch and I would like it to act as a normal switch (permanent on/permanent off behavior). How can this be achieved?

[image]

Jagomeister, 2018-07-02:
Create a new network input
Set Parameter:
Read value from --> Local variable memory (VM)

Set an address
eg
VB: 0
Bit: 0

in node-red setup the s7 endpoint
under variables set address to: DB1,X0.0

Banane84, 2018-07-16:
Hi I'm trying to conect my logo to red node is the readme by gfcittolin is done?
I dont know how configurate neither logo or red node side

Thanks

Jagomeister, 2018-07-17:
What are you trying to do read/set blocks or read flags/inputs/outputs

In Node Red:
Add Input s7 node
Edit node add new connection

Connection Tab
IP: ip of Logo
Mode: TSAP
Local TSAP: 01.00 
Remote TSAP: 02.00

Variables Tab
add variables using the vm address map from help 
eg for Q1
DB1,X1064.0

Q8
DB1,X1064.7

Jagomeister, 2018-07-20:
Local is 01:00 not 00:00
Remote is 02:00 not 20:00

Banane84, 2018-07-21:
Sorry I really don't understand nothing.. and I've not even a little debug message to help.. I tried many things from yesterday.. What I missed I a global view of how it works.. I'm really sorry

Banane84, 2018-07-21:
Thanks a lot I reached to:

- 

 with VM Adresses to write in logo  (a bit strange the dashboard buttons only set to 'true' i've to reset them to 'false' after)

 also to read my nlocs
I also reached with

Banane84, 2018-07-22:
Another question... to retreive one bit X one Byte INT 
and a double byte (word?)
Thanks

gfcittolin, 2018-07-23:
In Logo, all memory addresses that you can access from this node are mapped to a "Virtual Data Block", and this happens to be the DB number one, the DB1. This is all done inside Logo itself. 

So, if you have another Logo, you'll have another config (with different IP address, for example), but still using the DB1, because the Logo will internally do the translation to the address you actually want.

gfcittolin, 2018-07-23:
Btw, you can post example flows on https://flows.nodered.org

Banane84, 2018-07-24:
Thanks a lot for help I created a flow as adviced.. It's a first step all helps, comments will be welcome
https://flows.nodered.org/flow/272a6d2104474c68f7b0e01bae025ead

gfcittolin, 2018-07-24:
Awesome! It's always nice to have an example when we want to do something, so it's much appreciated :) Thanks for it!

Banane84, 2018-07-25:
Do you think it would be complicated to do directly a LOGO node.. I think you did the big of the job.. I could think about easier settings and configuration for LOGO..
But I don't know how is it working for maintenance when new releases of node7 happens
I also don't know how to declare a Byte in S7 endpoint variables
**Bit >> X0.0 X0.1 X0.2......**   Boolean Values
_Byte (8bit)>> ?0 ?2 ?3..._   Some block parameters values
**Word (16bit)>> INT0, INT2, INT4.....** Analogic Values
**Double Word (32bit)>>DINT0, DINT4, DINT8.....** not met by yet

gfcittolin, 2018-07-25:
What we've already thought about and discussed a bit was, instead of a new node for Logo, it would be already nice just to implement an address translation for it, e.g one would write `AI1`, and it would internally do the translation to `DB1,W1032`. This multiple translation thing is something we're planning to implement on the underlying nodeS7 library.

Regarding byte access, you can try `DB1,Bxxx`, where `xxx` is the address you want to access. But remember: you can only read the areas as listed on [this table](https://www.promotic.eu/en/pmdoc/Subsystems/Comm/PLC/LOGO.htm). So, for example, if you read `DB1,B1104`, you will be reading M1 to M8 from Logo in a single byte.

gfcittolin, 2018-07-25:
These are the data types and their respective byte-length that you can configure (the `y` part in `DB1,yxxx`):

 - 4 bytes: `REAL`, `DWORD`, `DINT`
 - 2 bytes: `INT`, `WORD`
 - 1 byte: `B`, `C`, `BYTE`, `CHAR`
 - single bit: `X` (then of course you need to address the bit itself with a `.`(dot) in the `xxx` part, e.g. `1024.0`)

Banane84, 2018-07-25:
Ok you mean I can write `INT` or `WORD` it would work the same? perhaps to have it clearer i could put `WORD` and `BYTE` intead of `INT` int the flow and the little speech to have it the same as in logo.

Yes I did understood that you can make mistakes of this type (I tried :-D )
I don't know s7, I just have logo..

But for LOGO it could be a lot more easy If you want one value you select first I Q AI AQ...VM 
- I Q AI... You chose the number... 
- VM you choose `WORD` `BYTE` `X` an the adress in the VM area
 it would make  all this much easier..

Would it be a problem to map all values all the time or it would be better to preserve bandwith/CPU to take only needed values

Banane84, 2018-07-25:
So in logo? I have to use `DINT` `INT` `BYTE` and `X` or??

Can I help somehow

gfcittolin, 2018-07-25:
I'm sorry, but it depends on what you're trying to read. If you're unsure, just stick to what that table in the link that Jagomeister and me and others mentioned before

Banane84, 2018-07-25:
All OK.. It's what I did, I just wanted to have the flow as good as possible.. It's already quite good.. And just for me it's all working rally good.. Somehow looks like values are signed in AI AM and VM because I used INT and  i've the good numbers rendered in plus and minus.. Again thaks a lot for the big job