st-one-io/node-red-contrib-s7 #6: Run flow_witeTest.json Error

hhdang, 2017-06-19:
Return follow Error after deploy flow_writeTest.json 
-------------------------------------------------------
TypeError: this is not a typed array.
    at Function.from (native)
    at NodeS7.onTCPConnect (/home/hh/.node-red/node_modules/nodes7/nodeS7.js:247:19)
    at Socket.<anonymous> (/home/hh/.node-red/node_modules/nodes7/nodeS7.js:163:21)
    at Socket.g (events.js:260:16)
    at emitNone (events.js:67:13)
    at Socket.emit (events.js:166:7)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1050:10)

gfcittolin, 2017-06-21:
Hi, this has been introduced with a PR we've made to implement the ability to connect to the PLC by using TSAP instead of Rack/Slot. The `Buffer.from()` is only available in NodeJS versions greater than 6.x. I've created a PR to fix this on plcpeople/nodeS7#34. Meanwhile, if possible, you can update your NodeJS version, and the issue won't happen

gfcittolin, 2017-06-22:
This was fixed upstream, going to close the issue

closed 2017-06-22 by gfcittolin