st-one-io/node-red-contrib-s7 #8: Reading from Analog Network Ouputs

HelloHungryImDad, 2017-07-06:
Firstly, thanks for making this!  It has been a great help. 

However, I can't seem to get any values from an analog network output. I can read digital just fine, and the node creates an object key for the analog variable but the values are always zero. Is this a limitation of this library or does the S7 protocol not support analog values?

HelloHungryImDad, 2017-07-06:
To provide additional information:

I'm using a Siemens logo 0ba8 with TSAP 02.00 on both PLC and node red, configured to use the S7 protocol. 

Im sending booleans to IB0, IB1, QB0, and QB1 which I can read from node red. 

I'm sending an analog value (word) to IW2 but no value is reported in node red, however an object is created in the output payload. 

So my output payload is {IB: <data>, QB: <data>, IW:0}.

gfcittolin, 2017-07-11:
Hi,

Unfortunately we don't have a Logo for testing here, but as far as I know the communication protocol should be the same. There's just a difference in the communication setup process, that we've implemented in the latest version

Maybe you can check this link https://www.promotic.eu/en/pmdoc/Subsystems/Comm/PLC/LOGO.htm on how to map the the addresses, so you could double-check they're being mapped correctly. There it says that the analog inputs start at 1032, while the analog memories start at 1118,

So for example, for reading the first analog input I would try configuring the address `DB1,W1032`. Which address have you configured for your variables?

HelloHungryImDad, 2017-07-14:
Hi,

My analog input is AI2, so I have mapped the output to DB2.DBW1032 as suggested by the documentation. However, now node-red does not even output an object for this. See screenshots below.

Logo configuration (192.168.0.254 is my node-red instance):
[image]

Connection config in node-red:
[image]

Variables config in node-red:
[image]

node-red output:
[image]

Rikki

gfcittolin, 2017-08-14:
Sorry for the long time to answer. The syntax we use doesn't match exactly the one in TIA Portal or Step7. Please try to config it exactly as I posted above. So, for the list in your screenshot above, you'd have:
```
pedals: DB1,B1024 (First input byte)
breakes: DB1,B1064 (First output byte
torque: DB1,W1034 (Second analog input word - AI2)
```
Note there is a comma (`,`) between `DB1` and the address part, and there's no `DB` between the comma and the address type.

Again, you can check the samples on the "S7 address" on the following link, that has a nice mapping table between Logo addresses and S7 protocol addresses: https://www.promotic.eu/en/pmdoc/Subsystems/Comm/PLC/LOGO.htm

You can also post your flow (JSON) to take another look at it

gfcittolin, 2017-08-25:
I'm closing this as this seems to be working. Let me know if it's not the case

closed 2017-08-25 by gfcittolin