st-one-io/node-red-contrib-s7 #9: s7-1200 with raspberry pi 2

burak8758, 2017-08-09:
Hello guys,

is this package working with raspberry pi? I installed this node and tried to communicate with s7-1200 but, ı couldn't send any data. connected LED which below of node is green during connected ethernet cable. But when I try send or receieve data, there is no action.

Could you help me please?

gfcittolin, 2017-08-14:
Hi @burak8758 

This package works normally on RPi. It may be the case you have issues with the addresses of your variables, as right now we still don't do any type of validation at configuration time.

Take a look at these examples (taken from https://github.com/plcpeople/nodeS7/blob/master/README.md)
```
TEST2: 'M32.2',        // Bit at M32.2
TEST1: 'MR4',          // Memory real at MD4
TEST3: 'M20.0',        // Bit at M20.0
TEST4: 'DB1,REAL0.20', // Array of 20 values in DB1
TEST5: 'DB1,REAL4',    // Single real value
TEST6: 'DB1,REAL8',    // Another single real value
TEST7: 'DB1,INT12.2'   // Two integer value array
```

If you still have problems you can post your flow or a screenshot of your config so we can take a look at it.
Just for knowing, are you trying to connect to the CPU directly or to a CP?

burak8758, 2017-08-15:
hello @gfcittolin  i am trying to connect to the CPU directly. I don't have CP module. I will try your directory. I haven't tried yet. Thanks a lot.

gfcittolin, 2017-08-20:
I'm closing the issue for now, as there's apparently no bug here. Feel free to share your results trying to configure it, and we can reopen it in case of bugs

closed 2017-08-20 by gfcittolin