st-one-io/node-red-contrib-s7 #14: Improvement: Ethernet devices

icarogregorio, 2017-09-06:
Hi,

I have a suggestion to improve the node configuration.
I need to include the select of ethernet device because there SBC with two or more interfaces!

Thanks a lot!

gfcittolin, 2017-09-08:
Hi @icarogregorio,

The node plays the role of TCP/IP Client here, without opening server sockets. Therefore, it doesn't make much sense to specify the ethernet interface, as the choice of the interface will be done by the OS based on the IP to connect to and the OS's routing table.

AFAIK there isn't even a way to specify this for client sockets on Node.JS. Could you please explain better your issue with the interfaces? Maybe the solution for it is at OS level

gfcittolin, 2018-03-02:
I'm closing this issue, as there's nothing much we can do here. Feel free to reopen it if needed.

closed 2018-03-02 by gfcittolin