st-one-io/node-red-contrib-s7 #15: TSAP Settings Not Available

alberto464, 2017-09-17:
Hello,

Tsap settings are not available, is there a way to turn them on?
The version of library is 1.3.1

[image]

gfcittolin, 2017-09-18:
That's weird, I see no way of hiding it. Also noticed the "Rack" icon is an old one, not the one we use on 1.3.x.
Maybe somehow the module was not updated, or then node-red was not restarted after the update, or even your browser is caching the resources. Could you check these points?

alberto464, 2017-09-18:
I installed the library yesterday morning, I don't know how fix the problem

gfcittolin, 2017-10-03:
Hi, I really couldn't simulate your issue. Are you still having this issue? Could you try a clean installation?

closed 2017-11-08 by gfcittolin