st-one-io/node-red-contrib-s7 #18: "Error connecting to PLC: Error: read ECONNRESET"

Spawn2qc, 2018-02-28:
I have communicate between S7-1200/S7-1500 and Node-RED on IOT2040 many times in the past.

Yesterday, I installed a brand new image V2.1.3 on a IOT2040, change IP address to my needs, auto-start Node-red, installed S7 node and finally reboot before doing my flow.

I usually use the "Rack/slot" mode (not TSAP) for the "s7 endpoint node" and access inputs/outputs and variables in DB without any issue.

Here is the current configuration to communicate with a S7-1200:

[image]
[image]

Here is the DB:
[image]

Yes, I check the checkbox "Permit access with PUT/GET communication from remote partner".
[image]

I can see my data is receive by the S7 node, but my debug window is fill with the error "Error connecting to PLC: Error: read ECONNRESET".
[image]

I never had this error in the past.

I receive my data, that's not the end of the world, but it is annoying to have a filled debug window.

I tried to add a S7 connection with an "Unknown partner" with IP address of the IOT2040 and use the TSAP mode, but get the same result. 

Do you have an idea what's wrong?

It must be related to the latest version of S7 node 1.3/1.4 because I never got this error in the past and I am doing the same thing.

gfcittolin, 2018-03-02:
Hi @Spawn2qc 

I think the issue might be with a connection leak we're having under some circumstances. In the case we have a connection problem, we currently ask the library to connect again, but without creating a new library instance, and then the connection leaks

It might be in your case that the PLC is not accepting connections anymore because of these leaked connection, and therefore it logs these `ECONNRESET` errors.

I've been testing the fix for this in the past weeks, and it looks like the problem is solved. I'll publish a new version either today or tomorrow, so you can update your node and check if the problem has been fixed

gfcittolin, 2018-03-02:
Just published 1.5.0 that might fix the issue. Could you please update and check it?

Spawn2qc, 2018-03-03:
Hi @gfcittolin 
I think you are right about the PLC not accepting connections anymore.
This morning, I reboot all the system and the S7-1200 must have reset its connection because there is no more error in the debug window.
Yesterday, I reboot Node-RED only, not the S7-1200.
So, maybe, the "leaked" connection in the S7-1200 were still count.
There is 8 S7 connections limit, maybe it was bust.
[image]
I will update S7 node to 1.5.0 monday morning and let you know if something goes wrong, else it is solve.
Thank you very much for your support, it is very appreciate!

gfcittolin, 2018-03-14:
My pleasure. Let me know if it works now so we can close this issue.

zoujagger, 2018-03-16:
Hi every body,
I am having exactly the same issue using the latest release of the node-red-contrib-s7 1.5.0. 
Indeed, in one hand i am getting the error : "Error connecting to PLC: error: read ECONNRESET" continuously and in the other hand, i receive the data well! However, after a certain period of time, the node-red crashes.  

Please find below the main configurations my architecture.

configuration:
PLC S7-1200
node red 0.18.4
node-red-contrib-s7 1.5.0
OS Ubuntu 16.0.4

Does anybody  have an idea or can help!? It's an emergency ... Thanks

Best regards

gfcittolin, 2018-03-16:
That is bad news. Could you please:
 - Check the connection resources of your S7-1200 and get the available connections, to confirm our hypothesis that we're leaking connections
 - Post here the stack trace of Node-RED when it crashes, so we can track down where it fails
 - Enable the debug mode on the PLC config on node-red, and then try to simulate the issue. Then get the logs of node-red and post here too, so we can analyze what's wrong.

gfcittolin, 2018-03-19:
Just released 1.5.1, there was another connection leak case I could simulate here. The leak happened when the node was trying to reconnect to the PLC, and the flow was redeployed. Could you guys please test it out?

zoujagger, 2018-03-20:
Thanks for your contribution, We are test it and we'll give you the result

zoujagger, 2018-03-21:
I tested the 1.5.1 release  yesterday, and everything works fine. I no longer have the error message. 
Thanks  for  your reactivity. 

Best regards

closed 2018-03-21 by Spawn2qc

gfcittolin, 2018-03-21:
Thanks for testing!

lukaku75, 2019-05-24:
Hi,
I have descripted problem above!
But I can't get data from my plc.
The node worked fine for one week but then I get these Errors:
[image]
I do several things:
-Reset the SPS
-uninstall the node and install it new
-upgrading node-red.
- I  checked the Ressources
[image]
-I checked the mechanism
[image]

but the ERROR is still there.
What else can I do?
Thanks!