st-one-io/node-red-contrib-s7 #20: s7-Out, send a Boolean command

mrsalhaoui, 2018-03-12:
Hello,
I'm trying to send the value true or false to my variable test in the PLC, using the node S7, but it doesn't work when I put the adress : DB3.X42.0 or even DB3.BOOL42.0.

[image]

gfcittolin, 2018-03-14:
Hi,
You should try DB3,X42.0 (with a comma instead of the first dot). We use here the naming schema of the library we use under the hood, [plcpeople/nodes7](https://github.com/plcpeople/nodeS7).

gfcittolin, 2018-03-14:
Oh, now I see you've used the comma on the screenshot, the dot was a typo on your text. 
Shouldn't your string "MatriculaEntrante" be `DB3,S10.30`, as it's 30 bytes long?

If you don't succeed, maybe could you activate the debug mode and post the logs from Node-RED, so we can try to understand why it's not working.

mrsalhaoui, 2018-03-14:
Yes, but the string works very well, and the number 32 for the string, came from the difference between the offset of the Bool variable and the string's offset, I tried yesterday to put the variable Bool before the variable string in the table, and it works, but when is directly after the string it doesnt !! why ? Can you show me please how to activate the debug mode ?

gfcittolin, 2018-03-14:
You can activate the debug mode in the config node of the PLC:
[image]

The bool access should indeed work either before or after the string, so there may be a bug here. But I'd suggest trying to set the string length to 30. The underlying library should account for the 2 extra bytes used to store the string length. The way you have it configured you may be asking to read beyond the memory area of your DB.

mrsalhaoui, 2018-03-15:
Yes , you are right, when I changed the string length to 30, it worked, the value 32 was like an excess of memory that is why I didn't receive anything in the next value. Thank you.
[image]
Would you please send to us all cases of variables that can we send to the PLC, and supported by the node S7.
Thank you.

gfcittolin, 2018-03-15:
We support reading all basic types from DBs (boolean, int, dint, word, dword, real, char, string), and additionaly timers and counters when reading from other memory areas.

We plan to add support to field validation in the next minor version of the node, but until there, the only way to be sure of the supported types is by looking at the sources on the library [here](https://github.com/plcpeople/nodeS7/blob/master/nodeS7.js#L1992)

closed 2018-03-16 by mrsalhaoui