st-one-io/node-red-contrib-s7 #23: Large number of variables

KarlSayle, 2018-04-17:
Hi all, 

We are having issues with NodeRed grinding to a halt when querying a large number of variables from the PLC. The variables are imported from a CSV. Is there a 'correct' method for handling large numbers of variables?

Thanks.

KarlSayle, 2018-04-17:
Hi, its around 2000 and NodeRed is currently running on an IOT2040. Just want to point out, it was never orignally the plan to have to pull out so many variables! Thanks.

gfcittolin, 2018-04-17:
Oh. Maybe it's a bit too much for the hardware. As far as I know, the CPU used on IOT2000 lacks the performance bits needed to run javascript with decent performance. A colleague of mine tried to read less than half of this using Node-RED on top of IOT2000 and had also performance issues.

A possible improvement would be to put all your variables inside a single DB, and read from there. The read of contiguous memory areas are optimized and may improve the performance there. Other than this, I can't think on anything else that would make some difference. If you decrease the amount of variables, does it work normally?

P.S.: We have applications at this scale, but running on our hardware, [ST-One](https://netsmarttech.com/page/st-one). It's a more powerful box running Node-RED out of the box.

KarlSayle, 2018-04-18:
Thanks for the feedback. The ST-One looks like a nice unit! Ill keep that in mind for next time.
Ill try cutting back on the variables and see what effect it has.
Thanks again.

closed 2018-05-18 by gfcittolin