st-one-io/node-red-contrib-s7 #25: Error: "Error connecting to PLC: Error: read ECONNRESET" with Version 1.5.1

DanielT2408, 2018-05-30:
Hello Ladies and Gentleman
I get the error "Error connecting to PLC: Error: read ECONNRESET" after I updated my node from 1.4.1 to version 1.5.1.
I get the error sometimes two times in one hour, and after certain period of time Node-Red restarts or crashes.

I tested the new version on my windows PC and Node-Red, there i have no problems. Is it possible, that there is still a problem with the version on Ubuntu Core ?

I´m using Ubuntu Core 16-2.32.6 on a Dell 3001 IoT - Gateway und a Node red version 0.17.5.

My questions:
1. Has anyone still the same problems with the version of 1.5.1?
2. Is it possible to install the old Version 1.4.1, and how can I install an older version?

Thank you for your answer
Best regards Daniel

gfcittolin, 2018-06-07:
Hello Daniel,

There's little chance that we have a problem on a specific OS. There's no platform-specific code, and it should work the same way independently of the OS. We use this node almost only on Linux boxes without further problems.

> I get the error sometimes two times in one hour, and after certain period of time Node-Red restarts or crashes.

Could you please post here the message it prints when Node-RED crash? So we can further investigate if there's an issue there. The PLCs can close the connection due to numerous reasons (network timeout, malformed/corrupted packets, not enough resources, etc.), but Node-RED should never crash and exit.

> Is it possible to install the old Version 1.4.1, and how can I install an older version?

Yes, you can revert to 1.4.1 by manually running `npm install node-red-contrib-s7@1.4.1` where your modules reside on your installation (typically `~/.node-red` on linuxes). But it would be better if we could fix the issue on 1.5.1, so you don't have to use old versions of this node.

gfcittolin, 2018-06-27:
Hi, does the issue persist? If it doesn't, let me know so we can close the issue

Schreck1985, 2018-08-01:
Hello,

we have a Windows Server 2008 R2 machine and a CP343 Lean which we are trying to connect to. But immediately after deploying the config, `ECONNRESET` appers permanently, and then the node opens another instance with port 102. After deleting the node and taking a look at the system monitor, we had about 4 sessions running and trying to connect to the same CP with port 102. We are runnig the latest version of your node, 1.5.1 .

If you tell me what to do, I can investigate further to help us solve the problem. We want to use node red and S7 for mailing alerts and stuff.

gfcittolin, 2018-08-01:
The system monitor's indication as you described is definitely an indication of a connection leak. I'll try to take a deeper look at the code, but it would be also great if you could simulate the exact moment when the leak happens (at flow deploy, at reconnect after failure, etc). 
I'm currently traveling on a project, so it'll be hard for me to spend too much time on it (and even harder to test with a real PLC), but I'll do my best :)

Schreck1985, 2018-08-02:
At first, a connection can be established after deploy, but the nodes argues about bad values (I tried to read 'DB999,INT1624', hope I wrote it properly). After trying around a little bit and restarting the flows, your node cannot connect to the CP anymore. I tried to paste some debug information.

```
2 Aug 07:28:14 - [info] [s7 endpoint:92b2ffb1.5d5fe8] Initiating connection to P
LC
[5789288,947991086] Initiate Called - Connecting to PLC with address and paramet
ers:
[5789288,948331702] { host: '192.168.203.100', port: '102', rack: '0', slot: '4'
 }
[5789288,948785880] Connection cleanup is happening
[5789288,949094787 192.168.203.100 S4] Clearing read PacketTimeouts
[5789288,949416197 192.168.203.100 S4] Clearing write PacketTimeouts
[5789288,949888534 192.168.203.100 S4] <initiating a new connection Thu Aug 02 2
018 07:28:14 GMT+0200 (Mitteleuropäische Sommerzeit)>
[5789288,950257855 192.168.203.100 S4] Attempting to connect to host...
[5789288,952560800 192.168.203.100 S4] TCP Connection Established to 192.168.203
.100 on port 102
[5789288,952904630 192.168.203.100 S4] Will attempt ISO-on-TCP connection
[5789288,953270947 192.168.203.100 S4] Using rack [0] and slot [4]
[5789288,957315106 192.168.203.100 S4] We Caught a connect error ECONNRESET
2 Aug 07:28:14 - [error] [s7 endpoint:92b2ffb1.5d5fe8] Error connecting to PLC:
Error: read ECONNRESET
[5789288,958672821] ConnectionReset is happening
[5789289,872237394 172.22.45.52 S3] Reading All Items (readAllItems was called)
[5789289,872651273 172.22.45.52 S3] Calling SRP from RAI
[5789289,873020734 172.22.45.52 S3] SendReadPacket called
[5789289,873478473 172.22.45.52 S3] Sending Read Packet
[5789289,884289623 172.22.45.52 S3] Received 6 bytes of S7-data from PLC.  Seque
nce number is 2
[5789289,884674240 172.22.45.52 S3] ReadResponse called
[5789289,885045725 172.22.45.52 S3] Time is 0 seconds and 11.7 ms.
[5789289,885375166 172.22.45.52 S3] Address DB200,INT0 has value 63 and quality
OK
[5789289,885953522 172.22.45.52 S3] We are calling back our readDoneCallback.
[5789290,466232720 192.168.203.100 S4] PacketTimeout called with type connect an
d seq undefined
[5789290,466745843 192.168.203.100 S4] TIMED OUT connecting to the PLC - Disconn
ecting
[5789290,467257012 192.168.203.100 S4] Wait for 2 seconds then try again.
[5789290,467605171] ConnectionReset is happening
[5789290,467945019 192.168.203.100 S4] Scheduling a reconnect from packetTimeout
, connect type
[5789290,468669412] ResetNOW is happening
[5789290,469102009] Clearing an earlier scheduled reset
[5789291,887808506 172.22.45.52 S3] Reading All Items (readAllItems was called)
[5789291,888178386 172.22.45.52 S3] Calling SRP from RAI
[5789291,888709459 172.22.45.52 S3] SendReadPacket called
[5789291,889210012 172.22.45.52 S3] Sending Read Packet
[5789291,903178197 172.22.45.52 S3] Received 6 bytes of S7-data from PLC.  Seque
nce number is 2
[5789291,903599061 172.22.45.52 S3] ReadResponse called
[5789291,903974877 172.22.45.52 S3] Time is 0 seconds and 14.9 ms.
[5789291,904303969 172.22.45.52 S3] Address DB200,INT0 has value 63 and quality
OK
[5789291,904724344 172.22.45.52 S3] We are calling back our readDoneCallback.
[5789292,481716112 192.168.203.100 S4] The scheduled reconnect from packetTimeou
t, connect type, is happening now
[5789292,482356138] Connection cleanup is happening
[5789292,482919827 192.168.203.100 S4] Clearing read PacketTimeouts
[5789292,483237046 192.168.203.100 S4] Clearing write PacketTimeouts
[5789292,483693388 192.168.203.100 S4] <initiating a new connection Thu Aug 02 2
018 07:28:18 GMT+0200 (Mitteleuropäische Sommerzeit)>
[5789292,484060195 192.168.203.100 S4] Attempting to connect to host...
[5789292,487081948 192.168.203.100 S4] TCP Connection Established to 192.168.203
.100 on port 102
[5789292,487415231 192.168.203.100 S4] Will attempt ISO-on-TCP connection
[5789292,487773377 192.168.203.100 S4] Using rack [0] and slot [4]
[5789292,492633562 192.168.203.100 S4] We Caught a connect error ECONNRESET
[5789292,492884710] ConnectionReset is happening
2 Aug 07:28:18 - [info] Stopping flows
2 Aug 07:28:18 - [debug] red/nodes/flows.stop : stopping flow : global
2 Aug 07:28:18 - [info] [s7 endpoint:92b2ffb1.5d5fe8] Closing connection to PLC
2 Aug 07:28:18 - [debug] red/nodes/flows.stop : stopping flow : 46ee4a8d.448c24
2 Aug 07:28:18 - [debug] red/nodes/flows.stop : stopping flow : 8f521ee6.4eb75
[5789292,856161328 172.22.45.52 S3] ISO-on-TCP connection DISCONNECTED.
[5789292,856398858] ConnectionReset is happening
[5789292,856896198] ConnectionReset is happening
[5789293,997195105 192.168.203.100 S4] PacketTimeout called with type connect an
d seq undefined
[5789293,997703549 192.168.203.100 S4] TIMED OUT connecting to the PLC - Disconn
ecting
[5789293,998187480 192.168.203.100 S4] Wait for 2 seconds then try again.
[5789293,998540038] ConnectionReset is happening
[5789293,998876533 192.168.203.100 S4] Scheduling a reconnect from packetTimeout
, connect type
[5789293,999404254] ResetNOW is happening
[5789293,999691162] Clearing an earlier scheduled reset
[5789294,356602166] ResetNOW is happening
[5789294,356979867] Clearing an earlier scheduled reset
[5789295,356695474] Connection cleanup is happening
[5789295,357124789 192.168.203.100 S4] Clearing read PacketTimeouts
[5789295,357343182 192.168.203.100 S4] Clearing write PacketTimeouts
2 Aug 07:28:20 - [info] Stopped flows
```

If you need further testing, don't hesitate to contact me.

Schreck1985, 2018-08-02:
Hi again,

in the meanwhile we tried other connections to other PLCs and they work like a charm. Either the CP has an error or the network switch connected to the CP. So maybe we have a hardware problem rather than a problem with your node ;-) but I will give you feedback if you wish to.

Schreck1985, 2018-08-02:
Ok, sorry. Obviously it was our fault. If you are trying to connect through the CP, you still have to name rack / slot of the PLC, not the CP. Otherwise the connection gets refused. For us the case is closed. Thank you for your answer!

closed 2018-10-21 by gfcittolin