st-one-io/node-red-contrib-s7 #26: Node connection

icarogregorio, 2018-07-17:
The node does not disconnect after we deactivate the flow that has a configuration node.
When this happens the log gets filled by it because the configuration node try to connect again when the timeout parameter reaches out

gfcittolin, 2018-10-09:
By default, the configuration nodes have a global scope, and so they cannot be disabled. If you want to disable it when disabling a flow, you can scope it to a specific flow, and then you it should stop when the this flow is disabled.

To change the configuration node's scope, you can change it here:
[image]

Just tested this here and looks like it works as described above.
Could you guys check out this, or have I misundestood something?

gfcittolin, 2018-10-29:
Hi, I could just simulate the issue. The problem seems to happen when the node is "undeployed" (closed) when the nodeS7 library is trying to do the reconnection. I'll investigate further and I'll post here as soon as I have more infos about that

gfcittolin, 2019-02-15:
Hi guys,
Just published version 1.5.4 pointing to a fresh version of nodes7 where a connection issue has been resolved. Hope it will fix this issue. Could you guys try it out?

felipedream, 2019-05-13:
Hello people, Very good work with this.
I have a problem with showing an INT data to a DB table.
I can read data
and I can only send True or false to an M0.0 memory.
It will be possible that PLCSIM ADV does not work very well?

[image]
[image]

gfcittolin, 2019-05-13:
According to your screenshots, the address you should configure is `DB2,I14`. The number 6 is only the row number and means nothing to the PLC or to Node-RED

closed 2019-06-05 by gfcittolin