st-one-io/node-red-contrib-s7 #30: LOGO! and Node-RED communications

tsangiotis, 2018-10-27:
I get the following behavior:

I connect to a Siemens LOGO! and read some inputs without a problem.

When I disconnect the LOGO! from power and reconnect after a while, I get only "Failure (Bad values)".

[image]

The only way to go back to a working situation is to do a Full Deploy or restart my Node-RED instance.

Here is the flow:

```
[{"id":"dceccca6.e4c29","type":"tab","label":"S7-LOGO8","disabled":false,"info":"First Step to adapt S7-node to logo-node\n\nLogoComfort side:\n\nTools-->Ethernet connections\nEnter:\n-LOGO IP adress\n-Subnet Mask\n-Default Gateway\n\nRight Click on Ethernet connections\nAdd Server connection --> S7 Connection\nDouble click on connection1\n\nTick Connect with an Operator Panel (OP)\nIt will change TSAP to 02.00\n\nTick Accept all connection requests (Or you know what you do)\n\nRemote Properties (Client) TSAP: 01.00\n\nRed Node Side:\nMenu (Upper Right)-->Import-->Clipboard\nPaste flow\nImport\n\nIn S7-LOGO8 Flow\nDouble click on LOGO8\nPLC properties (little pencil)\nChange for your Red Node IP adress\n\nDeploy\nNow you could already read all I/O of Logo\n"},{"id":"14e4c23a.cca3ee","type":"s7 in","z":"dceccca6.e4c29","endpoint":"7a40424c.fd2d5c","mode":"single","variable":"I01 AI03","diff":true,"name":"LOGO8 I1","x":120,"y":40,"wires":[["a924a0e8.3f8a3"]]},{"id":"5083fbb4.8ce664","type":"s7 in","z":"dceccca6.e4c29","endpoint":"7a40424c.fd2d5c","mode":"single","variable":"I02 AI04","diff":true,"name":"LOGO8 I2","x":120,"y":100,"wires":[["afec231c.166fd"]]},{"id":"5696677b.25aba8","type":"s7 in","z":"dceccca6.e4c29","endpoint":"7a40424c.fd2d5c","mode":"single","variable":"I03 High Freq","diff":true,"name":"LOGO8 I3","x":120,"y":160,"wires":[["4d88bfd7.616db"]]},{"id":"63ce8ec9.fb9a4","type":"s7 in","z":"dceccca6.e4c29","endpoint":"7a40424c.fd2d5c","mode":"single","variable":"I04 High Freq","diff":true,"name":"LOGO8 I4","x":120,"y":220,"wires":[["7375958c.ee97fc"]]},{"id":"37357f81.c71a1","type":"s7 in","z":"dceccca6.e4c29","endpoint":"7a40424c.fd2d5c","mode":"single","variable":"I05 High Freq","diff":true,"name":"LOGO8 I5","x":120,"y":280,"wires":[["c43162c3.6c6d8"]]},{"id":"d37c85c7.e78558","type":"s7 in","z":"dceccca6.e4c29","endpoint":"7a40424c.fd2d5c","mode":"single","variable":"I06 High Freq","diff":true,"name":"LOGO8 I6","x":120,"y":340,"wires":[["b920f6d6.ea8ae8"]]},{"id":"ad4ba9ca.cfe948","type":"s7 in","z":"dceccca6.e4c29","endpoint":"7a40424c.fd2d5c","mode":"single","variable":"I07 AI01","diff":true,"name":"LOGO8 I7","x":120,"y":400,"wires":[["a5a96891.a05988"]]},{"id":"2699a98d.1f0a66","type":"change","z":"dceccca6.e4c29","name":"","rules":[{"t":"change","p":"payload","pt":"msg","from":"true","fromt":"bool","to":"1","tot":"str"},{"t":"change","p":"payload","pt":"msg","from":"false","fromt":"bool","to":"0","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":540,"y":40,"wires":[["a80acc0.95afd38","4ce79648.3e4278"]]},{"id":"afec231c.166fd","type":"change","z":"dceccca6.e4c29","name":"","rules":[{"t":"set","p":"topic","pt":"msg","to":"2","tot":"num"}],"action":"","property":"","from":"","to":"","reg":false,"x":290,"y":100,"wires":[["2699a98d.1f0a66"]]},{"id":"4d88bfd7.616db","type":"change","z":"dceccca6.e4c29","name":"","rules":[{"t":"set","p":"topic","pt":"msg","to":"3","tot":"num"}],"action":"","property":"","from":"","to":"","reg":false,"x":290,"y":160,"wires":[["2699a98d.1f0a66"]]},{"id":"7375958c.ee97fc","type":"change","z":"dceccca6.e4c29","name":"","rules":[{"t":"set","p":"topic","pt":"msg","to":"4","tot":"num"}],"action":"","property":"","from":"","to":"","reg":false,"x":290,"y":220,"wires":[["2699a98d.1f0a66"]]},{"id":"c43162c3.6c6d8","type":"change","z":"dceccca6.e4c29","name":"","rules":[{"t":"set","p":"topic","pt":"msg","to":"5","tot":"num"}],"action":"","property":"","from":"","to":"","reg":false,"x":290,"y":280,"wires":[["2699a98d.1f0a66"]]},{"id":"b920f6d6.ea8ae8","type":"change","z":"dceccca6.e4c29","name":"","rules":[{"t":"set","p":"topic","pt":"msg","to":"6","tot":"num"}],"action":"","property":"","from":"","to":"","reg":false,"x":290,"y":340,"wires":[["2699a98d.1f0a66"]]},{"id":"a5a96891.a05988","type":"change","z":"dceccca6.e4c29","name":"","rules":[{"t":"set","p":"topic","pt":"msg","to":"7","tot":"num"}],"action":"","property":"","from":"","to":"","reg":false,"x":290,"y":400,"wires":[["2699a98d.1f0a66"]]},{"id":"a80acc0.95afd38","type":"function","z":"dceccca6.e4c29","name":"format arguments","func":"msg.payload = String(msg.topic) + \" \" + String(msg.payload)\nreturn msg;","outputs":1,"noerr":0,"x":630,"y":120,"wires":[["6bec640.c14fa9c","4ce79648.3e4278"]]},{"id":"6bec640.c14fa9c","type":"exec","z":"dceccca6.e4c29","command":"cd  ~/maintnode/ && pipenv run python maintnode/logo/iomeasure.py","addpay":true,"append":"","useSpawn":"false","timer":"","oldrc":false,"name":"commit measurement","x":760,"y":200,"wires":[["d1bb3e2b.9176a"],[],["d1bb3e2b.9176a"]]},{"id":"a924a0e8.3f8a3","type":"change","z":"dceccca6.e4c29","name":"","rules":[{"t":"set","p":"topic","pt":"msg","to":"1","tot":"num"}],"action":"","property":"","from":"","to":"","reg":false,"x":290,"y":40,"wires":[["2699a98d.1f0a66"]]},{"id":"d1bb3e2b.9176a","type":"debug","z":"dceccca6.e4c29","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":870,"y":280,"wires":[]},{"id":"4ce79648.3e4278","type":"debug","z":"dceccca6.e4c29","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":810,"y":40,"wires":[]},{"id":"bfad1c08.5174b","type":"s7 out","z":"dceccca6.e4c29","endpoint":"7a40424c.fd2d5c","variable":"M03","name":"Unlock","x":630,"y":520,"wires":[]},{"id":"bb76c64a.2f7cc8","type":"inject","z":"dceccca6.e4c29","name":"","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":150,"y":520,"wires":[["bfad1c08.5174b","db928faa.7f1fa"]]},{"id":"5fc82588.a341ac","type":"s7 in","z":"dceccca6.e4c29","endpoint":"7a40424c.fd2d5c","mode":"single","variable":"M03","diff":true,"name":"LOGO8 M3","x":130,"y":460,"wires":[["80d59433.659b48"]]},{"id":"80d59433.659b48","type":"debug","z":"dceccca6.e4c29","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":370,"y":460,"wires":[]},{"id":"db928faa.7f1fa","type":"delay","z":"dceccca6.e4c29","name":"","pauseType":"delay","timeout":"1","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":300,"y":580,"wires":[["2621c6b5.34718a"]]},{"id":"2621c6b5.34718a","type":"change","z":"dceccca6.e4c29","name":"","rules":[{"t":"set","p":"payload","pt":"msg","to":"false","tot":"bool"}],"action":"","property":"","from":"","to":"","reg":false,"x":460,"y":580,"wires":[["bfad1c08.5174b"]]},{"id":"7a40424c.fd2d5c","type":"s7 endpoint","z":"","address":"192.168.1.122","port":"102","rack":"0","slot":"2","localtsaphi":"01","localtsaplo":"00","remotetsaphi":"02","remotetsaplo":"00","connmode":"tsap","cycletime":"500","timeout":"1500","verbose":"off","name":"S7-LOGO8","vartable":[{"addr":"DB1,X1064.0","name":"Q01"},{"addr":"DB1,X1064.1","name":"Q02"},{"addr":"DB1,X1064.2","name":"Q03"},{"addr":"DB1,X1064.3","name":"Q04"},{"addr":"DB1,X1064.4","name":"Q05"},{"addr":"DB1,X1064.5","name":"Q06"},{"addr":"DB1,X1064.6","name":"Q07"},{"addr":"DB1,X1064.7","name":"Q08"},{"addr":"DB1,X1065.0","name":"Q09"},{"addr":"DB1,X1065.1","name":"Q10"},{"addr":"DB1,X1065.2","name":"Q11"},{"addr":"DB1,X1065.3","name":"Q12"},{"addr":"DB1,X1065.4","name":"Q13"},{"addr":"DB1,X1065.5","name":"Q14"},{"addr":"DB1,X1065.6","name":"Q15"},{"addr":"DB1,X1065.7","name":"Q16"},{"addr":"DB1,X1066.0","name":"Q17"},{"addr":"DB1,X1066.1","name":"Q18"},{"addr":"DB1,X1066.2","name":"Q19"},{"addr":"DB1,X1066.3","name":"Q20"},{"addr":"DB1,X1024.0","name":"I01 AI03"},{"addr":"DB1,X1024.1","name":"I02 AI04"},{"addr":"DB1,X1024.2","name":"I03 High Freq"},{"addr":"DB1,X1024.3","name":"I04 High Freq"},{"addr":"DB1,X1024.4","name":"I05 High Freq"},{"addr":"DB1,X1024.5","name":"I06 High Freq"},{"addr":"DB1,X1024.6","name":"I07 AI01"},{"addr":"DB1,X1024.7","name":"I08 AI02"},{"addr":"DB1,X1025.0","name":"I09"},{"addr":"DB1,X1025.1","name":"I10"},{"addr":"DB1,X1025.2","name":"I11"},{"addr":"DB1,X1025.3","name":"I12"},{"addr":"DB1,X1025.4","name":"I13"},{"addr":"DB1,X1025.5","name":"I14"},{"addr":"DB1,X1025.6","name":"I15"},{"addr":"DB1,X1025.7","name":"I16"},{"addr":"DB1 [... cut ...] 1,INT1120","name":"AM2"},{"addr":"DB1,INT1122","name":"AM3"},{"addr":"DB1,INT1124","name":"AM4"},{"addr":"DB1,INT1126","name":"AM5...64"},{"addr":"DB1,X1246.0","name":"NI1"},{"addr":"DB1,X1246.1","name":"NI1..64"},{"addr":"DB1,X1390.0","name":"NQ1"},{"addr":"DB1,X1390.1","name":"NQ1..64"},{"addr":"DB1,INT1262","name":"NAI1"},{"addr":"DB1,INT1264","name":"NAI1...64"},{"addr":"DB1,INT1406","name":"NAQ1"},{"addr":"DB1,INT1408","name":"NAQ2...64"}]}]
```

Any ideas on how to get a stable connection?

marioishikawa, 2018-11-21:
I am seeing the same issue. It stays on failure state for an undetermined time. Did you get any solution to this @tsangiotis ?

tsangiotis, 2018-11-22:
I observed the same issue with the Modbus node red module so I believe it has something to do with an underlying library.

I ended up using the node red daemon module along with https://github.com/riptideio/pymodbus in a simple script.

gfcittolin, 2018-11-26:
Hi guys,

The node should always reconnect automatically. There's currently another open issue (#26) regarding a connection being kept open, that is caused by the underlying communication library. 
In the case of the S7 family, this shouldn't be an issue, as long as there are enough resources available on the PLC. In the case of Logo!, as we manually set the TSAP values, it may reject the connection because there's already one with the same parameters. 

To confirm this hypothesis, could one of you please simulate the issue with the "Debug" option set to "On", and then post here the Node-RED logs of it?

marioishikawa, 2018-12-19:
> Hi guys,
>
> The node should always reconnect automatically. There's currently another open issue (#26) regarding a connection being kept open, that is caused by the underlying communication library. [...]

I don't have access to one now but I'll do the test in probably 3 weeks.

nymJ, 2018-12-27:
I run Node-Red in a docker container. After I few times I got an error with my LOGO! connection:

`[200704,681450500 192.168.178.222 S2] Address DB1,X1106.3 has value false and quality OK
[200704,681467400 192.168.178.222 S2] We are calling back our readDoneCallback.
[200704,864147100 192.168.178.222 S2] Preparing to WRITE M01 to value true
[200704,864490500 192.168.178.222 S2] Clearing write PacketTimeouts
[200704,864864800 192.168.178.222 S2] Sending Write Packet With Sequence Number 621
[200704,867190500 192.168.178.222 S2] Received 1 bytes of S7-data from PLC.  Sequence number is 621
[200704,867406600 192.168.178.222 S2] Time is 0 seconds and 2.8 ms.
[200704,867516500 192.168.178.222 S2] DB1,X1104.0 write completed with quality OK
[200705,176608600 192.168.178.222 S2] Reading All Items (readAllItems was called)
[200705,176914600 192.168.178.222 S2] Calling SRP from RAI
[200705,177003100 192.168.178.222 S2] SendReadPacket called
[200705,177913800 192.168.178.222 S2] Sending Read Packet
[200705,181274400] INVALID READ RESPONSE - DISCONNECTING
[200705,181492500] TPKT Length From Header is 42 and RCV buffer length is 41 and COTP length is 2 and data[6] is 128
[200705,181527700] <Buffer 03 00 00 2a 02 f0 80 32 03 00 00 00 03 00 02 00 15 00 00 04 03 ff 04 00 20 08 43 16 16 ff 04 00 08 00 00 ff 04 00 18 01 00>
[200705,181703700] ConnectionReset is happening
27 Dec 00:12:08 - [red] Uncaught Exception:
27 Dec 00:12:08 - RangeError: Index out of range
    at checkOffset (buffer.js:831:11)
    at Buffer.readInt16BE (buffer.js:975:5)
    at checkRFCData (/data/node_modules/nodes7/nodeS7.js:1448:27)
    at NodeS7.onResponse (/data/node_modules/nodes7/nodeS7.js:1094:11)
    at Socket.<anonymous> (/data/node_modules/nodes7/nodeS7.js:370:20)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at readableAddChunk (_stream_readable.js:176:18)
    at Socket.Readable.push (_stream_readable.js:134:10)
    at TCP.onread (net.js:559:20)

npm ERR! Linux 4.9.125-linuxkit
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "start" "--" "--userDir" "/data"
npm ERR! node v6.14.4
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! node-red-docker@1.0.0 start: `node $NODE_OPTIONS node_modules/node-red/red.js -v $FLOWS "--userDir" "/data"`npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-red-docker@1.0.0 start script 'node $NODE_OPTIONS node_modules/node-red/red.js -v $FLOWS "--userDir" "/data"'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the node-red-docker package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node $NODE_OPTIONS node_modules/node-red/red.js -v $FLOWS "--userDir" "/data"
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs node-red-docker
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls node-red-docker
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /usr/src/node-red/npm-debug.log`

marioishikawa, 2019-01-11:
@gfcittolin on Debug panel, it comes only:
Failure (Bad values)

gfcittolin, 2019-02-15:
Hi guys,
Just published version 1.5.4 pointing to a fresh version of nodes7 where a connection issue has been resolved. Hope it will fix this issue. Could you try it out?

Sascha-Leadbetter, 2019-05-21:
We are using the 1.6.0 and also get the same error:

nodered_1          | [106045,280937200 192.168.0.1 S2] ReadResponse called
nodered_1          | [106045,281001000 192.168.0.1 S2] Time is 0 seconds and 261.5 ms.
nodered_1          | [106045,281056100 192.168.0.1 S2] Address DB61,INT42 has value 1 and quality OK
nodered_1          | [106045,281076100 192.168.0.1 S2] Address DB61,INT44 has value 9 and quality OK
nodered_1          | [106045,281085900 192.168.0.1 S2] Address DB61,INT46 has value 36 and quality OK
nodered_1          | [106045,281101600 192.168.0.1 S2] Address DB61,CHAR48.72 has value LS 44.2     1.WT nicht vorhanden(3x schlecht)                            and quality OK
nodered_1          | [106045,281111400 192.168.0.1 S2] Address DB61,DINT134 has value 0 and quality OK
nodered_1          | [106045,281134700 192.168.0.1 S2] We are calling back our readDoneCallback.
nodered_1          | [106045,338518300 192.168.0.3 S2] Reading All Items (readAllItems was called)
nodered_1          | [106045,338623500 192.168.0.3 S2] Calling SRP from RAI
nodered_1          | [106045,338637500 192.168.0.3 S2] SendReadPacket called
nodered_1          | [106045,338720500 192.168.0.3 S2] Sending Read Packet
nodered_1          | [106045,338735700 192.168.0.1 S2] Reading All Items (readAllItems was called)
nodered_1          | [106045,338758000 192.168.0.1 S2] Calling SRP from RAI
nodered_1          | [106045,338765000 192.168.0.1 S2] SendReadPacket called
nodered_1          | [106045,338811900 192.168.0.1 S2] Sending Read Packet
nodered_1          | [106045,596579600] INVALID READ RESPONSE - DISCONNECTING
nodered_1          | [106045,596628000] TPKT Length From Header is 39 and RCV buffer length is 38 and COTP length is 2 and data[6] is 128
nodered_1          | [106045,596656600] <Buffer 03 00 00 27 02 f0 80 32 03 00 00 00 02 00 02 00 12 00 00 04 02 ff 04 00 30 00 01 00 09 00 24 ff 04 00 20 00 00 00>
nodered_1          | [106045,596692800] ConnectionReset is happening
nodered_1          | 16 May 15:57:58 - [red] Uncaught Exception:
nodered_1          | 16 May 15:57:58 - RangeError: Index out of range
nodered_1          |     at checkOffset (buffer.js:977:11)
nodered_1          |     at Uint8Array.Buffer.readInt16BE (buffer.js:1121:5)
nodered_1          |     at checkRFCData (/usr/src/node-red/node_modules/nodes7/nodeS7.js:1474:27)
nodered_1          |     at NodeS7.onResponse (/usr/src/node-red/node_modules/nodes7/nodeS7.js:1120:11)
nodered_1          |     at Socket.<anonymous> (/usr/src/node-red/node_modules/nodes7/nodeS7.js:390:20)
nodered_1          |     at emitOne (events.js:116:13)
nodered_1          |     at Socket.emit (events.js:211:7)
nodered_1          |     at addChunk (_stream_readable.js:263:12)
nodered_1          |     at readableAddChunk (_stream_readable.js:250:11)
nodered_1          |     at Socket.Readable.push (_stream_readable.js:208:10)
nodered_1          |     at TCP.onread (net.js:601:20)
nodered_1          | npm ERR! code ELIFECYCLE
nodered_1          | npm ERR! errno 1
nodered_1          | npm ERR! node-red-docker@1.0.0 start: `node $DEBUG_OPTIONS $NODE_OPTIONS node_modules/node-red/red.js -v $FLOWS "--userDir" "/data" "--settings" "/usr/src/node-red/settings.js"`
nodered_1          | npm ERR! Exit status 1
nodered_1          | npm ERR!
nodered_1          | npm ERR! Failed at the node-red-docker@1.0.0 start script.
nodered_1          | npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
nodered_1          |     at readableAddChunk (_stream_readable.js:250:11)
nodered_1          |     at Socket.Readable.push (_stream_readable.js:208:10)
nodered_1          |     at TCP.onread (net.js:601:20)

We could also provide wireshark logs. But for us it would be best if in case of an error this exception is catched and our system doesn't crash.

gfcittolin, 2019-05-21:
I don't know how I missed that on the logs of @nymJ, but there's indeed another issue, not related to the connection problem we had before `v.1.6.0`. This one is also, albeit a different one, is also caused by the underlying library.
 
On the Buffer printed out, we can calculate that the last byte is missing, and that's why it drops the connection in the first place. The bigger issue then happens when we receive the missing byte in another packet, that is, a single byte telegram. The `nodeS7` library is definitely not prepared for such a case, and an exception is thrown there.

I'm opening an issue there, so we can work on a fix for it.

gfcittolin, 2019-05-21:
@Sascha-Leadbetter could you please post here (or send me privately if you prefer) the wireshark dumps, so I can confirm that we're receiving a single-byte telegram, and then test with the fixes?

marioishikawa, 2019-05-24:
> Hi guys,
> Just published version 1.5.4 pointing to a fresh version of nodes7 where a connection issue has been resolved. Hope it will fix this issue. Could you try it out?

Hi. I just tested with 1.6.0 and I still get the same error:

`[71307,665421342 172.22.105.30 S2] Reading All Items (readAllItems was called)
[71307,665725611 172.22.105.30 S2] Calling SRP from RAI
[71307,665833788 172.22.105.30 S2] SendReadPacket called
[71307,666134099 172.22.105.30 S2] Sending Read Packet
[71309,171022086 172.22.105.30 S2] PacketTimeout called with type read and seq 2
[71309,171467501 172.22.105.30 S2] READ TIMEOUT on sequence number 2
[71309,171726042 172.22.105.30 S2] ReadResponse called
[71309,171890885] Processing an undefined packet, likely due to timeout error
[71309,172094061] Timeout error - zero length packet
[71309,172262967 172.22.105.30 S2] Received a ZERO RESPONSE Processing Read Packet due to unrecoverable packet error
[71309,172448956 172.22.105.30 S2] Time is 1 seconds and 506.5 ms.
[71309,172737705 172.22.105.30 S2] Address DB50,DINT0 has value 0 and quality BAD 255
[71309,172937391 172.22.105.30 S2] Address DB50,DINT4 has value 0 and quality BAD 255
[71309,173141557 172.22.105.30 S2] Address DB50,DINT8 has value 0 and quality BAD 255
[71309,173316400 172.22.105.30 S2] Address DB50,DINT12 has value 0 and quality BAD 255
[71309,173488014 172.22.105.30 S2] Address DB50,DINT16 has value 0 and quality BAD 255
[71309,173677545 172.22.105.30 S2] Address DB50,DINT20 has value 0 and quality BAD 255
[71309,173830148 172.22.105.30 S2] We are calling back our readDoneCallback.
24 May 08:28:05 - [error] [s7 endpoint:1f492ce5.3a5b13] Failure (Bad values)`

gfcittolin, 2019-06-05:
What exactly is the error you're reporting? As far as I understood, there's the first issue with the node not being able to reconnect. and another one that crashed node-red.
The first one should be fixed, while the second one has a partial fix, preventing Node-RED to crash, but still dropping the connection (and connecting again automatically thereafter.

If possible, please send the whole logs of node-red with the issue, as I unfortunately don't have a Logo here for testing. Thanks!

marioishikawa, 2019-07-01:
> What exactly is the error you're reporting? As far as I understood, there's the first issue with the node not being able to reconnect. and another one that crashed node-red.
> The first one should be [...]

Hi, my last message was regarding the first issue, with Node-Red not reconnecting. I'll repeat the test and see what other logs I can send you. Thank you!

gfcittolin, 2020-03-24:
Is this still an issue after the improvements done on nodes7? I'm closing this issue for now since the last activity was almost one year ago, but feel free to reopen it and let me know if there's still any kind of connectivity issue.

closed 2020-03-24 by gfcittolin