st-one-io/node-red-contrib-s7 #33: Emit only when value changes not working with array of bits

AlexanderLL95, 2018-12-08:
Hi,

I think this is a new issue.

I found out how to read an array of bits from a PLC.
This is actually working. I get the correct values.

I use the following address to read the array of bits: DB60,X0.0.100

I checked the node property 'Emit only when value changes (diff)'.

This has no effect. The node ignores this setting.
The values are readed cyclic.

When i just ask for a specific bit. The setting is working.
See the screenshot of the debug output.

See the other screenshots for more details.

**Node configuration**
[image]

**Flow**
[image]

**Debug output**
[image]

gfcittolin, 2018-12-10:
Great catch! Indeed, we just check for plain javascript equality (`==`), and for arrays it won't work, as javascript will try to compare the references and will obviously fail (as they're not the same object, but may have the same content)
I'll try to fix it asap and report here so you can test it.

AlexanderLL95, 2018-12-10:
Hi gfcittolin,
Thanks for the quick respons.
I'm looking forward for this fix.

gfcittolin, 2019-01-17:
Just published 1.5.3 with the fix for that, sorry for the huge delay. Let me know if you have any other issue

commit bfe31d66 (2019-01-17): Fixes #33: handle diff of array types
closed 2019-01-17 by gfcittolin