st-one-io/node-red-contrib-s7 #35: Sending string

MrBoggi, 2019-01-21:
I'm trying to send a string from Node Red to my S-1200 CPU.
I have a DB configured with both real, bool and strings defined. Real and bool works og but strings give me an error.
21.1.2019, 13:29:38node: PLC
msg : error
"TypeError: "value" argument is out of bounds"

PLC node config:
[image]

My plc config:
[image]

My inject node for test:
[image]

What can be the issue here?

gfcittolin, 2019-01-21:
Your string seems to be too big. Strings have a single byte on their start to store the max. length, and another byte to store the current length, limiting the size of a string to 0~255 characters.

Also keep in mind that these 2 bytes must be accounted for when calculating the space needed. If you have 256 bytes available (304.0 - 48.0 = 256), then your string can have a max length of 254. Therefore, your tag should be `DB600,S48.254`.

Another option is to use a char array instead of a string, so you don't have this limitation. But then you have to keep track of used bytes elsewhere, and handle this both on Node-RED and in your PLC program

MrBoggi, 2019-01-21:
Well.. That makes sense.. and fixes my problem. Hmmm... Thanks for quick help :)

closed 2019-01-21 by MrBoggi

gfcittolin, 2019-01-21:
Great, you're welcome!