st-one-io/node-red-contrib-s7 #36: works on PLCSIM but only get "bad values" on a real S7-1200

jupiter8888, 2019-03-04:
Well, i will need a hand on this one,

every thing works o.k in simulation with PLCSIM,  so i changed the ethernet adress, complied, uploaded program in a s7-1200, started everything, tested o.k with TIA portal, can see values changing in my watch table and yes, put/get is enable, block optimisation disable.

 I'm running Node-Red locally in the same laptop as TIA. 
 put the plc's ethernet address in the node. Without any variable the node is connected and no errors are reported. but with a variable i get "bad data" in the debug tab and "Error S7 endpoint : xxx Error connecting to PLC [object object]" in node-red debug window. 

Nothing is changed, only ethernet address, same flow used for simulation.
Node-red V.0..19.6 
S7 node V.1.5.4

any idea what can cause this?

Thanks.

jupiter8888, 2019-03-07:
Finally nailed it. It looks like installed firmware version 4.1 was the culprit, afler installing version 4.2 it works.

But beware, i could'nt start the cpu after because of version mismatch, you have to replace the cpu from the catalog and reupload program and redownload it in TIA to make it work.

gfcittolin, 2019-03-10:
I've already heard stories about fixing connection problems by upgrading the firmware, but it wouldn't cross my mind as a suggestion for this. Great that it worked after all, and thanks for sharing the solution!

closed 2019-03-10 by gfcittolin