st-one-io/node-red-contrib-s7 #37: Control node: trigger read doesn't work

AlbeToff, 2019-03-25:
Hi, why if the input node is on "Emit only when change" mode, and  if I want read the input "manually" with the control node the trigger doesn't work?

Thank you

gfcittolin, 2019-03-25:
I don't know if I understood your issue correctly. The expected behavior is that, when "Emit only on change" is checked, the "s7 in" node will emit only if the observed value has changed, regardless of the trigger method, be that automatically or manually triggered with the control node. If you want to receive a message everytime you manually trigger a read process, then you need to uncheck that checkbox.
Are you experiencing a different behavior?

AlbeToff, 2019-03-25:
So isn't possible keep checked "Emit only on change" mode, but at the same time  manually trigger the the input with a control node for resend the msg.payload from the input node to a function even if the entrance has not changed?

gfcittolin, 2019-03-25:
That functionality is currently not supported. Despite of that, there are many ways you could have such a logic in the flow itself, by using change, switch and rbe nodes

AlbeToff, 2019-03-25:
Ok Thanks!

closed 2019-03-25 by AlbeToff