st-one-io/node-red-contrib-s7 #38: 1.6.0 @ node-red v0.20.5 in docker container: extensive writing to log file

MaFoB, 2019-04-16:
If I add just one variable to be read from the S7 an extensive logging to file starts with more than 20 lines per second such as
_{"log":"[1542040,726948427 172.23.156.112 S2] SendReadPacket called\n","stream":"stdout","time":"2019-04-16T14:27:08.550057364Z"}
{"log":"[1542040,727169084 172.23.156.112 S2] Sending Read Packet\n","stream":"stdout","time":"2019-04-16T14:27:08.550175097Z"}_
see file attached.
[node-red-contrib-s7_1.6.0_logging.txt](https://github.com/netsmarttech/node-red-contrib-s7/files/3085394/node-red-contrib-s7_1.6.0_logging.txt)
=> Disk full and system blocked soon. Can you help?

gfcittolin, 2019-04-16:
The official Node-RED container starts with Node-RED in verbose mode, which in turn causes the massive amount of logs. 
You can manually disable the node's logging on the "s7 endpoint" config node. There's a debug option there that you can set to "No", and the logs should then stop.

MaFoB, 2019-04-17:
Thanks for your quick response. - "Debug Off" works indeed :)
(However, [error]s are still logged.)

[image]
_Issue already mentioned here:_
_Originally posted by @gfcittolin in https://github.com/netsmarttech/node-red-contrib-s7/issues/12#issuecomment-330108836_

Our problem still is "... writing to log file":
In our node-red docker implementation all command line output is also written to a file.
Well, we have to solve this docker specific part ourselves. Just would be glad if you have any advice...

gfcittolin, 2019-04-17:
This options just en/disable the logging of the library we use to communicate with (since it currently cannot integrate with a logging framework). For all other logging, you should be able to control that globally on your `settings.js` of Node-RED.

Regarding Docker logging, it all depends how you've set-up your environment, but a quick look at the Docker Docs ([here](https://docs.docker.com/config/containers/logging/configure/) and [here](https://docs.docker.com/config/containers/logging/json-file/)) shows that you can configure the default logging driver to do log rotation automatically. Maybe that can help.

MaFoB, 2019-04-17:
Yes, it does help us docker freshers a lot. Thanks !

closed 2019-04-17 by gfcittolin