st-one-io/node-red-contrib-s7 #39: LOGO/S7-200 - Write value to Digital Output (Q...)

Brmm, 2019-04-28:
Hi, 

Not sure if this applies to all S7 versions but I have a Logo 8.2 (0BA8). 

Is it possible to write a boolean to the Q1 (DB1,X1064.0) address? I have been successful on writing booleans to flag addresses e.g. M01 (DB1,X1104.0) but I can't seem to flip the switch on Digital Outputs. 

The thing is, I can read it perfectly but no matter what I write like boolans, numbers or strings I don't get any errors or any results.

Logs:
```
Preparing to WRITE Q01 to value true
DB1,X1064.0 write completed with quality OK
Address DB1,X1064.0 has value false and quality OK
```

Flow:
```
[{"id":"3136edb4.c805d2","type":"tab","label":"Flow 1","disabled":false,"info":""},{"id":"738f6c54.dd8ec4","type":"s7 out","z":"3136edb4.c805d2","endpoint":"45a3399b.7baad8","variable":"Q01","name":"","x":670,"y":300,"wires":[]},{"id":"cb1f06b4.bd5868","type":"s7 out","z":"3136edb4.c805d2","endpoint":"45a3399b.7baad8","variable":"M01","name":"","x":670,"y":380,"wires":[]},{"id":"8c5f6766.768df8","type":"inject","z":"3136edb4.c805d2","name":"","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":330,"y":320,"wires":[["738f6c54.dd8ec4"]]},{"id":"24bc865d.e0da2a","type":"inject","z":"3136edb4.c805d2","name":"","topic":"","payload":"false","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":330,"y":380,"wires":[["738f6c54.dd8ec4"]]},{"id":"45a3399b.7baad8","type":"s7 endpoint","z":"","address":"192.168.100.50","port":"102","rack":"0","slot":"2","localtsaphi":"01","localtsaplo":"00","remotetsaphi":"02","remotetsaplo":"00","connmode":"tsap","cycletime":"500","timeout":"1500","verbose":"on","name":"LOGO","vartable":[{"addr":"DB1,X1064.0","name":"Q01"},{"addr":"DB1,X1064.1","name":"Q02"},{"addr":"DB1,X1064.2","name":"Q03"},{"addr":"DB1,X1064.3","name":"Q04"},{"addr":"DB1,X1064.4","name":"Q05"},{"addr":"DB1,X1064.5","name":"Q06"},{"addr":"DB1,X1064.6","name":"Q07"},{"addr":"DB1,X1064.7","name":"Q08"},{"addr":"DB1,X1065.0","name":"Q09"},{"addr":"DB1,X1065.1","name":"Q10"},{"addr":"DB1,X1065.2","name":"Q11"},{"addr":"DB1,X1065.3","name":"Q12"},{"addr":"DB1,X1065.4","name":"Q13"},{"addr":"DB1,X1065.5","name":"Q14"},{"addr":"DB1,X1065.6","name":"Q15"},{"addr":"DB1,X1065.7","name":"Q16"},{"addr":"DB1,X1066.0","name":"Q17"},{"addr":"DB1,X1066.1","name":"Q18"},{"addr":"DB1,X1066.2","name":"Q19"},{"addr":"DB1,X1066.3","name":"Q20"},{"addr":"DB1,X1024.0","name":"I01 AI03"},{"addr":"DB1,X1024.1","name":"I02 AI04"},{"addr":"DB1,X1024.2","name":"I03 High Freq"},{"addr":"DB1,X1024.3","name":"I04 High Freq"},{"addr":"DB1,X1024.4","name":"I05 High Freq"},{"addr":"DB1,X1024.5","name":"I06 High Freq"},{"addr":"DB1,X1024.6","name":"I07 AI01"},{"addr":"DB1,X1024.7","name":"I08 AI02"},{"addr":"DB1,X1025.0","name":"I09"},{"addr":"DB1,X1025.1","name":"I10"},{"addr":"DB1,X1025.2","name":"I11"},{"addr":"DB1,X1025.3","name":"I12"},{"addr":"DB1,X1025.4","name":"I13"},{"addr":"DB1,X1025.5","name":"I14"},{"addr":"DB1,X1025.6","name":"I15"},{"addr":"DB1,X1025.7","name":"I16"},{"addr":"DB1,X1026.0","name":"I17"},{"addr":"DB1,X1026.1","name":"I18"},{"addr":"DB1,X1026.2","name":"I19"},{"addr":"DB1,X1026.3","name":"I20"},{"addr":"DB1,X1026.4","name":"I21"},{"addr":"DB1,X1026.5","name":"I22"},{"addr":"DB1,X1026.6","name":"I23"},{"addr":"DB1,X1026.7","name":"I24"},{"addr":"DB1,INT1032","name":"AI01"},{"addr":"DB1,INT1034","name":"AI02"},{"addr":"DB1,INT1036","name":"AI03"},{"addr":"DB1,INT1038","name":"AI04"},{"addr":"DB1,INT1040","name":"AI06"},{"addr":"DB1,INT1042","name":"AI07"},{"addr":"DB1,INT1044","name":"AI08"},{"addr":"DB1,INT1046","name":"AI09"},{"addr":"DB1,INT1072","name":"AQ01"},{"addr":"DB1,INT1074","name":"AQ02"},{"addr":"DB1,INT1076","name":"AQ03"},{"addr":"DB1,INT1078","name":"AQ04"},{"addr":"DB1,INT1080","name":"AQ05"},{"addr":"DB1,INT1082","name":"AQ06"},{"addr":"DB1,INT1084","name":"AQ07"},{"addr":"DB1,INT1086","name":"AQ08"},{"addr":"DB1,INT1088","name":"AQ09"},{"addr":"DB1,X1104.0","name":"M01"},{"addr":"DB1,X1104.1","name":"M02"},{"addr":"DB1,X1104.2","name":"M03"},{"addr":"DB1,X1104.3","name":"M04"},{"addr":"DB1,X1104.4","name":"M05"},{"addr":"DB1,X1104.5","name":"M06"},{"addr":"DB1,X1104.6","name":"M07"},{"addr":"DB1,X1104.7","name":"M08 Start"},{"addr":"DB1,X1105.0","name":"M09"},{"addr":"DB1,X1105.1","name":"M10"},{"addr":"DB1,X1105.2","name":"M11"},{"addr":"DB1,X1105.3","name":"M12"},{"addr":"DB1,X1105.4","name":"M13"},{"addr":"DB1,X1105.5","name":"M14"},{"addr":"DB1,X1105.6","name":"M15"},{"addr":"DB1,X1105.7","name":"M16"},{"addr":"DB1,X1105.0","name":"M17"},{"addr":"DB1,X1105.1","name":"M18"},{"addr":"DB1,X1105.2","name":"M19"},{"addr":"DB1,X1105.3","name":"M20"},{"addr":"DB1,X1105.4","name":"M21"},{"addr":"DB1,X1105.5","name":"M22"},{"addr":"DB1,X1105.6","name":"M23"},{"addr":"DB1,X1105.7","name":"M24"},{"addr":"DB1,X1106.0","name":"M25 LOGO Screen White"},{"addr":"DB1,X1106.1","name":"M26 TDE Screen White"},{"addr":"DB1,X1106.2","name":"M27 CharSet"},{"addr":"DB1,X1106.3","name":"M28 LOGO Screen Orange"},{"addr":"DB1,X1106.4","name":"M29 LOGO Screen Red"},{"addr":"DB1,X1106.5","name":"M30 TDE Screen Orange"},{"addr":"DB1,X1106.6","name":"M31 TDE Screen Red"},{"addr":"DB1,X1106.7","name":"M32"},{"addr":"DB1,X1106.0","name":"M33...64"},{"addr":"DB1,INT1118","name":"AM1"},{"addr":"DB1,INT1120","name":"AM2"},{"addr":"DB1,INT1122","name":"AM3"},{"addr":"DB1,INT1124","name":"AM4"},{"addr":"DB1,INT1126","name":"AM5...64"},{"addr":"DB1,X1246.0","name":"NI1"},{"addr":"DB1,X1246.1","name":"NI1..64"},{"addr":"DB1,X1390.0","name":"NQ1"},{"addr":"DB1,X1390.1","name":"NQ1..64"},{"addr":"DB1,INT1262","name":"NAI1"},{"addr":"DB1,INT1264","name":"NAI1...64"},{"addr":"DB1,INT1406","name":"NAQ1"},{"addr":"DB1,INT1408","name":"NAQ2...64"}]}]
```

Thanks

gfcittolin, 2019-04-29:
Hi,

I think there shouldn't be any problem writing to outputs like this. Although I've never tested it on a Logo, this works normally on a 315 and on a 1212 I have here.
Just note that there must be nothing else writing on the outputs on the logic. If you're trying to set and output that the PLC logic is also writing to, the PLC will "overwrite" it. Maybe this is the case?

gfcittolin, 2019-04-29:
Just saw your edit. I've never put the hands on a Logo, so I can't help much on that.
That said, there's some useful info on the Snap7 project about connecting to a Logo (http://snap7.sourceforge.net/logo.html). We use the same protocol here, so the instructions should be roughly the same

Brmm, 2019-04-29:
@gfcittolin you are correct, thank you!

Solution is to have nothing connected to the output in the LOGO software. Best to completely remove it from the diagram. However it does not appear to work the same way for inputs sadly. If no input is present in the diagram it is not triggered physically (I'm testing with a pushbutton). Not connecting the input to anything in the diagram also gives an error when uploading the program to the LOGO controller. Solution I found there is to connect the input to a flag.

Regards

closed 2019-04-29 by Brmm