st-one-io/node-red-contrib-s7 #40: debug output is continuing after deletion of a deployed node-red-contib-s7

MaFoB, 2019-04-30:
I permanently experience a strange behavior of node-red-contrib-s7 after deleting both(!) the deployed node and the connection in Node-RED. It continues with the debug output as “ghost-node”. After restarting Node-RED, there is no debug output anymore (as expected).

Do you know about this issue? Is it related to node-red-contrib-s7 itself or maybe to the underlying library? Where should I post my observations (screenshots, debug output) to find a solution?

gfcittolin, 2019-04-30:
Hi, you can post your findings here, so we can analyze and determine the culprit. Besides the debug output and screenshots, please also post your flow and the version of node-red-contrib-s7 you're using, so we can better analyze it. Thanks!

MaFoB, 2019-05-14:
By now, I could not reproduce this effect anymore. Could be that is was due to https://github.com/netsmarttech/node-red-contrib-s7/issues/34#issuecomment-489085836

MaFoB, 2019-05-22:
No news are good news:
Restarting Node-RED in such cases of strange behavior seems to be the way to go.
(see also https://github.com/netsmarttech/node-red-contrib-s7/issues/41#issuecomment-494832809)

closed 2019-05-22 by MaFoB