st-one-io/node-red-contrib-s7 #41: "Error connecting to PLC" disturbing on Node-RED debug window if too many

MaFoB, 2019-05-03:
In my application, many PLC connections are deployed.
A large number of PLCs is regularly switched off => connection "offline". In this case, the large number of messages "Error connecting to PLC" floods the Node-RED debug window, however, which is not dependent on the flow but on the connections. Thus, messages from debug nodes cannot be read anymore by the user.

How about a debug option "Off and no connection errors" at the connection properties ?
(The info "online/connecting/offline" ist still visible as green/yellow/red indicator at the node.)

gfcittolin, 2019-05-13:
Yes, this could be a possibility. We could have a `Quiet` or `Silent` mode where all debug and warn messages would be suppressed (even though I fear about people complaining about the node not working, when in fact the error messages are not being shown).
Before we go for this option, have you tried changing the log levels on Node-RED itself?

gfcittolin, 2019-05-13:
I was checking it out here, and there's an easier solution for that. You can use the `Catch` node on Node-RED to get the errors emitted by other nodes. Just tested it here, and you can suppress the error messages thrown because the PLC is offline.

MaFoB, 2019-05-14:
Thank you for your answer. How do you suppress? Error is not thrown from node "s7 in" in the flow but from the configuration node in the background which still throws - see attachments:
[image]
[image]

gfcittolin, 2019-05-14:
Try defining your config nodes to the flow scope, they come on the "All flows" (global) scope by default. You can configure that on the bottom right corner of the s7 endpoint configuration

MaFoB, 2019-05-14:
Well, config nodes are just grouped then, "s7 endpoints" still throwing errors the same way, however...
[image]

MaFoB, 2019-05-14:
... but - as I have experienced in the meantime - not in other flows anymore :)
> This is part of the solution, indeed.
> However, I still do not understand whether/how the Catch node can help in the flow where the error belongs to.

gfcittolin, 2019-05-16:
When there's a Catch node catching errors thrown by other nodes, they're not shown on the debug window. This is the behavior I have (and that's what I've expected you'd have too)
[image]

MaFoB, 2019-05-22:
I found out that after deployment of the Catch node in a flow I need to restart Node-Red to have the behavior as shown and expected.
Before it is like this:
[image]

Thank you very much for your support !

closed 2019-05-22 by MaFoB