st-one-io/node-red-contrib-s7 #42: Uncaught Exception

corn98, 2019-06-03:
hello,
i am currently using the S7 node. Lately, i run into uncaught exception, more specifically
i get the following somehow randomly

[red] Uncaught Exception:
TypeError: Cannot read property '0' of undefined
    at bufferizeS7Item (/usr/lib/node_modules/node-red-contrib-s7/node_modules/nodes7/nodeS7.js:1921:55)
    at NodeS7.prepareWritePacket (/usr/lib/node_modules/node-red-contrib-s7/node_modules/nodes7/nodeS7.js:665:3)
    at NodeS7.writeItems (/usr/lib/node_modules/node-red-contrib-s7/node_modules/nodes7/nodeS7.js:455:7)
    at writeNext (/usr/lib/node_modules/node-red-contrib-s7/red/s7.js:239:28)
    at NodeS7.onWritten [as writeDoneCallback] (/usr/lib/node_modules/node-red-contrib-s7/red/s7.js:223:13)
    at NodeS7.writeResponse (/usr/lib/node_modules/node-red-contrib-s7/node_modules/nodes7/nodeS7.js:1276:8)
    at NodeS7.onResponse (/usr/lib/node_modules/node-red-contrib-s7/node_modules/nodes7/nodeS7.js:1186:10)
    at Socket.<anonymous> (/usr/lib/node_modules/node-red-contrib-s7/node_modules/nodes7/nodeS7.js:390:20)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)

and node-red crashes..
do you have any ideas on how solve this?

s7-node: v 1.6.0
Node-RED version: v0.20.3
Node.js version: v6.9.2

gfcittolin, 2019-06-03:
Hi,

From your stack trace, it looks like this is happening when you're writing a value to the PLC. It looks like you have a tag that is an Array, and you're trying to write a non-Array value to it. Regardless of whether this is the case, it should never crash Node-RED, but maybe it can help preventing it to happen meanwhile we don't have a fix

Could you provide your flow, so I can simulate the issue?

gfcittolin, 2019-06-05:
Yes, probably. I guess there's something wrong in your flow (although this is no reason for it to cause node-red to crash). If you can provide your flow, maybe we can check what type of data is causing it, so we can try to put safeguards preventing it to happen

corn98, 2019-06-05:
Hi,

I checked my flow in these day and I can run it surely for a work day (10h) without Node-Red crushes. 

The issue of "Failure (Bad values)" still remains but Node-Red not crushes. I think that those errors are related to the overlapping of readings/writings from/into the PLC.  Indeed the status indicator (below the nodes) shows 'Failure' for less then a second and then returns 'online'.

The problem is that, when I come back in office the next day, Node-Red was crushed (in the night I suppose) but I haven't the log for this situation. 

Tomorrow I'll keep switched on the PC and I update you for this issue.

Thanks for the support!

corn98, 2019-06-13:
I lightened my flow removing some writings/readings and the issue never happens anymore, Node-Red remains active.

If the issue occurs again I'll inform you with better details.

Corn

closed 2019-06-13 by corn98