st-one-io/node-red-contrib-s7 #43: Uncaught Exception at writeAfterEnd

viwi2005, 2019-06-03:
Hi thanks for this great plugin. During a test with a PLC connected via an LTE modem, our node-red constantly crashes due to the following exception: 

3 Jun 14:10:18 - [red] Uncaught Exception:
3 Jun 14:10:18 - Error [ERR_STREAM_WRITE_AFTER_END]: write after end
    at writeAfterEnd (_stream_writable.js:237:12)
    at Socket.Writable.write (_stream_writable.js:288:5)
    at Socket.write (net.js:710:40)
    at NodeS7.onISOConnectReply (/home/user/.node-red/node_modules/nodes7/nodeS7.js:335:17)
    at Socket.<anonymous> (/home/user/.node-red/node_modules/nodes7/nodeS7.js:291:26)
    at Socket.emit (events.js:159:13)
    at addChunk (_stream_readable.js:265:12)
    at readableAddChunk (_stream_readable.js:252:11)
    at Socket.Readable.push (_stream_readable.js:209:10)
    at TCP.onread (net.js:598:20)

To be honest, the LTE connection was very bad during the test. With a good and stable connection the problem doesn't occur.  

7-node:  1.6.0
Node-RED: 0.20.5
Node.js: 9.2.0

gfcittolin, 2019-06-05:
Hi,
Thanks for the report. I've checked the code and this seems to be indeed a very tricky condition. Looks like that somehow the connection had terminated from the Node.JS point of view, but even though we've received the ISO connection reply *after* the termination. Network can be a complex beast under some circumstances :/

I think there's a small check we can do to prevent this kind of thing to happen. This issue is actually from the underlying library, therefore I'm transferring the issue to the corresponding project.

gfcittolin, 2019-07-09:
Upstream bug has been fixed and new version is currently enforced on version 1.6.1

closed 2019-07-09 by gfcittolin