st-one-io/node-red-contrib-s7 #44: "extended" version of the S7 Protocol

PietroCasotto, 2019-07-04:
Hi!
Is there a way to implement the "extended" version of the S7 Protocol in this code?
I'm asking that because I'm recieving data from an s7-1500 from the internet and I can't make it works.
The problem is that I don't have access to the PLC that's sending me the data i need so i can't and don't want to reduce the security of the packet, for obvious reasons.
Thanks for your time

gfcittolin, 2019-07-04:
Hi,
This is am old wish we also have. We don't implement the protocol on this repository, but on the [plcpeople/nodeS7](https://github.com/plcpeople/nodeS7). The original codebase of that library would make it very difficult to extend, and therefore there's a major rewrite of that library going on (under the branch "next", done mostly by me). The initial focus is to replace the current functionality, but the basis for the "extended" protocol (a.k.a. user messages) is already there. If you have the knowledge/ability to contribute to that code let me know, that would be very appreciated.

Security note: even the "extended" version of that protocol is far from being "secure enough" to run over public internet. I'd advise to either have a tunneling/VPN, or to have an intermediate device by the PLC, handling the data acquisition and forwarding it by other better means.

PietroCasotto, 2019-07-04:
Hi! 
Thank you for your answer. 
When i read about this issue, the first thing I thougt was just as you are saying bout the library, that it desn't exist yet... Looking around the internet I found nothing about anything else. Unfortunately I don't know anything about the protocol, so I can't help... Thank you for your hard work.
I already have my PLC that sends the data to me through a VPN, so it's not a very big issue, but i was just asking if there was another solution...

closed 2019-07-04 by PietroCasotto