st-one-io/node-red-contrib-s7 #45: Lot of Variables

serd91000, 2019-07-09:
Hi,

I tried to import Variables and their address in CSV file. there are approximativly 20 000 variables. it is possible with your node. because When I tried, your node Collect the data of 4 or 5 variables and there was not the correct result.
My question : what is the correct way to collect a large number of variable? I make a wireshark capture to see the packet and data send to plc for collecting.

gfcittolin, 2019-07-09:
Hi,

There shouldn't be any limit on how many variables you're able to read. The only thing is that the larger the amount of variables (and their sizes), the longer it will take to read all of them, and therefore you can't have cycle times shorter than this amount of time. 

I didn't understand, what does happen when you put only a few variables? Does it work? One thing that may happen is, if there's any error on any variable (wrong address, syntax, etc.), the whole reading cycle will fail. So maybe you can double-check your variable list for such mistakes

gfcittolin, 2020-03-24:
Closing due to lack of activity. Please feel free to reopen the issue if needed

closed 2020-03-24 by gfcittolin