st-one-io/node-red-contrib-s7 #48: Uncaught Exception when variable is not selected

machadotiago, 2019-08-06:
It is possible to drop an output node without selecting the assigned variable to be written
[image]

This sometimes crashes Node-RED 

```
> ago 06 10:19:04.030074 st-one node-red[2411]: 6 Aug 10:19:04 - [red] Uncaught Exception:
> ago 06 10:19:04.035885 st-one node-red[2411]: 6 Aug 10:19:04 - TypeError: Cannot read property 'split' of undefined
> ago 06 10:19:04.035885 st-one node-red[2411]:     at stringToS7Addr (/home/node-red/node_modules/nodes7/nodeS7.js:2043:21)
> ago 06 10:19:04.035885 st-one node-red[2411]:     at NodeS7.writeItems (/home/node-red/node_modules/nodes7/nodeS7.js:433:35)
> ago 06 10:19:04.035885 st-one node-red[2411]:     at writeNext (/home/node-red/node_modules/node-red-contrib-s7/red/s7.js:239:28)
> ago 06 10:19:04.035885 st-one node-red[2411]:     at NodeS7.onWritten [as writeDoneCallback] (/home/node-red/node_modules/node-red-contrib-s7/red/s7.js:223:13)
> ago 06 10:19:04.035885 st-one node-red[2411]:     at NodeS7.writeResponse (/home/node-red/node_modules/nodes7/nodeS7.js:1285:8)
> ago 06 10:19:04.035885 st-one node-red[2411]:     at NodeS7.onResponse (/home/node-red/node_modules/nodes7/nodeS7.js:1195:10)
> ago 06 10:19:04.035885 st-one node-red[2411]:     at Socket.<anonymous> (/home/node-red/node_modules/nodes7/nodeS7.js:390:20)
> ago 06 10:19:04.035885 st-one node-red[2411]:     at Socket.emit (events.js:189:13)
> ago 06 10:19:04.035885 st-one node-red[2411]:     at addChunk (_stream_readable.js:284:12)
> ago 06 10:19:04.035885 st-one node-red[2411]:     at readableAddChunk (_stream_readable.js:265:11
```

gfcittolin, 2019-08-06:
Having a `s7 out` node without a selected variable is actually a feature. It enables you dynamically choose which variable to write to by using `msg.variable`.

I could however simulate the issue. It happens when trying to write to a variable that is not listed in the "Variables" list of the endpoint. Fixing it should be simple, I'll work on that.

gfcittolin, 2019-08-06:
Fixed in 2.0.1

commit c396fd9e (2019-08-06): Fixes #48 - prevents writing to unknown variables
closed 2019-08-06 by gfcittolin