st-one-io/node-red-contrib-s7 #50: No connection with S7-1512 st

rjboer, 2019-08-25:
We have a total of 7 plcs running with the excelent node-red-contrib-s7 package, 
This one is however different:

TIA portal:
DB optimization setting is off.
Access setting on full access
PUT end Get enabled. 

Settings node Red:
Rack /slot:= 0/1 (which is the setting in Tia).

It looks like we cannot find the db??

LOG:
[222039,529908519] ConnectionReset is happening
25 Aug 10:43:50 - [info] [s7 endpoint:96a52d88.76ca5] Initiating connection to PLC
[222039,532011826] Initiate Called - Connecting to PLC with address and parameters:
[222039,533918788] { host: '50.50.50.1', port: '102', rack: '0', slot: '1' }
[222039,534969651] Connection cleanup is happening
[222039,537882834 50.50.50.1 S1] Clearing read PacketTimeouts
[222039,539727771 50.50.50.1 S1] Clearing write PacketTimeouts
[222039,540636807 50.50.50.1 S1] <initiating a new connection Sun Aug 25 2019 10:43:50 GMT+0200 (W. Europe Daylight Time)>
[222039,541507917 50.50.50.1 S1] Attempting to connect to host...
[222039,544455076 50.50.50.1 S1] TCP Connection Established to 50.50.50.1 on port 102
[222039,545251124 50.50.50.1 S1] Will attempt ISO-on-TCP connection
[222039,545978037 50.50.50.1 S1] Using rack [0] and slot [1]
[222039,546876407 50.50.50.1 S1] ISO-on-TCP Connection Confirm Packet Received
[222039,551317687 50.50.50.1 S1] Received PDU Response - Proceeding with PDU 960 and 3 max parallel connections.
[222039,552963118] Translation OK
[222040,58285781 50.50.50.1 S1] Reading All Items (readAllItems was called)
[222040,60015755 50.50.50.1 S1] Adding or Removing { arg: [ 'testval' ], action: 'add' }
[222040,60651013 50.50.50.1 S1] Adding testval
[222040,62747999] Failed to find a match for DB44
[222040,63738418 50.50.50.1 S1] Dropping an undefined request item.
[222040,63943850 50.50.50.1 S1] Calling SRP from RAI
[222040,64536047 50.50.50.1 S1] SendReadPacket called
[222041,32880961] ResetNOW is happening

rjboer, 2019-08-25:
Wireshark trace
[wireshark trace.zip](https://github.com/netsmarttech/node-red-contrib-s7/files/3537969/wireshark.trace.zip)

Screenshot of the wireshark trace
[image]

rjboer, 2019-08-25:
Currently trying this with one variable. 
DB44.DBW0  <-db 44 is our reporting Database, dubbel word 0 -> 32bit (2x16) value

I tried the following rack/slot configs:
0/0
0/1 
0/2 <- makes no sense, but sure

gfcittolin, 2019-08-25:
Thanks a lot for your detailed report, this helps so much. From your logs, the item you're trying to read seems to be the culprit, probably its syntax. The wireshark dump confirms this, as the connection is successfully done, but no items are requested, so it's not the PLCs fault.

How have you written down this item? For that double word, the syntax should be `DB44,DW0` or `DB44,DWORD0`. Could you please confirm it, or then provide the flow so I can test it?

rjboer, 2019-08-26:
That fixed it, sorry to have wasted your time!

closed 2019-08-26 by rjboer