st-one-io/node-red-contrib-s7 #51: s7-200  Error: read ECONNRESET

lixinxiaobai, 2019-09-02:
I connected S7 200 PLC, suggesting Error connecting to PLC: Error: read ECONNRESET。
Who can help me?
windows10
"node-red": "^0.20.7",
"node-red-contrib-s7": "^1.6.3"

gfcittolin, 2019-09-02:
Hi, could you please provide a bit more details about it like what are you using to connect to the PLC, (CP, PPI-USB adapter, ethernet gateways, ...) and what is your configuration? Please provide your flow. Also, have you configured the connection at the PLC side (needed for S7-200 and Logo)?

lixinxiaobai, 2019-09-02:
thank you

I've already solved that, because I've been unsuccessful with slot before, and I don't know what happened.

Then I tried TSAP mode to solve the problem

lixinxiaobai, 2019-09-02:
But how to write data to PLC? I don't know how to use it

tstockner, 2019-10-17:
I have a similar problem.
I try to connect to my S7-300 via Ethernet on Rack 0 Slot 2 or 3. On Slot 2 is my CPU and on Slot 3 my Communication Point. Both fails with ECONNRESET and the big problem is that node-red crashes with it and with every restart of node-red it crashes again, so I can not apply the changes easily.

My setup looks like that:
S7-300 with IP 192.168.200.1
Host PC where a VM and node-red runs with IP 192.168.200.9
VM where TIA-Portal runs with IP 192.168.200.10

Nodejs-version: 10.16.3
npm-version: 6.9.0

I am pretty sure, that there is no problem with the virtual machine thing, cause I can watch the variables via TIA Portal

tstockner, 2019-10-17:
If someone else comes over ECONNRESET again, for me it was the wrong Portnumber.

gfcittolin, 2020-03-24:
Closing this issue due to lack of activity

closed 2020-03-24 by gfcittolin