st-one-io/node-red-contrib-s7 #52: Crashing on writing to variable

kempmaritz, 2019-09-13:
Ever since upgrade to v2.0.2 the node crashes Node-red when I write a value to the PLC.
I have an S7-315 with ethernet card (343-1)

Here is a snippet of the code that causes the crash.
[{"id":"54fcea12.a18494","type":"s7 out","z":"5ab44d6c.a3a794","endpoint":"6ed0dd77.8cc744","variable":"BackdoorLight_Cmd","name":"","x":730,"y":640,"wires":[]},{"id":"b578f24a.7e53b","type":"function","z":"5ab44d6c.a3a794","name":"On","func":"//Bit0\nvar status = 0;\nvar mask = 1;\nvar value;\nif (msg.payload === true)\n{\n    value = status | mask;\n}\nelse\n{\n    value = status & ~mask;\n}\nmsg.payload = value;\nreturn msg;","outputs":1,"noerr":0,"x":510,"y":640,"wires":[["54fcea12.a18494"]]},{"id":"245dea6.aeff316","type":"function","z":"5ab44d6c.a3a794","name":"Off","func":"//Bit1\nvar status = 0;\nvar mask = 2;\nvar value;\n\nif (msg.payload === true)\n{\n    value = status | mask;\n}\nelse\n{\n    value = status & ~mask;\n}\nmsg.payload = value;\nreturn msg;","outputs":1,"noerr":0,"x":510,"y":660,"wires":[["54fcea12.a18494"]]},{"id":"bf315410.591568","type":"function","z":"5ab44d6c.a3a794","name":"Manual","func":"//Bit3\nvar status = 0;\nvar mask = 8;\nvar value;\n\nif (msg.payload === true)\n{\n    value = status | mask;\n}\nelse\n{\n    value = status & ~mask;\n}\nmsg.payload = value;\nreturn msg;","outputs":1,"noerr":0,"x":520,"y":700,"wires":[["54fcea12.a18494"]]},{"id":"181a374b.c5ac99","type":"function","z":"5ab44d6c.a3a794","name":"Auto","func":"//Bit2\nvar status = 0;\nvar mask = 4;\nvar value;\n\nif (msg.payload === true)\n{\n    value = status | mask;\n}\nelse\n{\n    value = status & ~mask;\n}\nmsg.payload = value;\nreturn msg;","outputs":1,"noerr":0,"x":510,"y":680,"wires":[["54fcea12.a18494"]]},{"id":"1aa33712.dee4a9","type":"inject","z":"5ab44d6c.a3a794","name":"","topic":"","payload":"","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":250,"y":640,"wires":[["b578f24a.7e53b"]]},{"id":"7d30387d.e1f9f8","type":"inject","z":"5ab44d6c.a3a794","name":"","topic":"","payload":"","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":250,"y":660,"wires":[["245dea6.aeff316"]]},{"id":"95bf2645.96c868","type":"inject","z":"5ab44d6c.a3a794","name":"","topic":"","payload":"","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":250,"y":680,"wires":[["181a374b.c5ac99"]]},{"id":"1d6f321f.354f3e","type":"inject","z":"5ab44d6c.a3a794","name":"","topic":"","payload":"","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":250,"y":700,"wires":[["bf315410.591568"]]},{"id":"6ed0dd77.8cc744","type":"s7 endpoint","z":"","transport":"iso-on-tcp","address":"192.168.1.246","port":"102","rack":"0","slot":"2","localtsaphi":"01","localtsaplo":"00","remotetsaphi":"01","remotetsaplo":"00","connmode":"rack-slot","adapterauto":true,"adapterport":"","busaddr":"2","adapteraddr":"0","cycletime":"500","timeout":"1500","verbose":"default","name":"","vartable":[{"addr":"DB105,WORD24","name":"Sensors"},{"addr":"DB105,WORD2","name":"PoolPump_Status"},{"addr":"DB105,WORD8","name":"SecurityLight_Status"},{"addr":"DB105,WORD10","name":"BackdoorSolar_Status"},{"addr":"DB105,REAL12","name":"Poolpump_Current"},{"addr":"DB105,REAL16","name":"SolarVoltage"},{"addr":"DB26,BYTE50","name":"PoolPump_Command"},{"addr":"DB22,BYTE148","name":"SecLight_Command"},{"addr":"DB22,BYTE216","name":"BackSolar_Command"},{"addr":"DB22,BYTE420","name":"FairyLight_Command"}]}]

Can you help with this please?
Was working perfectly for the last year.

gfcittolin, 2019-09-13:
I've imported your flow and the variable you're writing to is not listed in the "Variables" tab (changed its name probably?). This shouldn't crash Node-RED, but the flow at least won't work.

Even though, could you please provide the logs of Node-RED when the problem happens?

kempmaritz, 2019-09-14:
I'm getting the following after I write a value to the PLC:

14 Sep 07:44:16 - ReferenceError: outputlog is not defined
    at NodeS7.readResponse (/home/snipe/.node-red/node_modules/nodes7/nodeS7.js:1427:4)
    at NodeS7.onResponse (/home/snipe/.node-red/node_modules/nodes7/nodeS7.js:1229:9)
    at Socket.<anonymous> (/home/snipe/.node-red/node_modules/nodes7/nodeS7.js:400:20)
    at emitOne (events.js:116:13)
    at Socket.emit (events.js:211:7)
    at addChunk (_stream_readable.js:263:12)
    at readableAddChunk (_stream_readable.js:250:11)
    at Socket.Readable.push (_stream_readable.js:208:10)
    at TCP.onread (net.js:607:20)

gfcittolin, 2020-03-24:
This was an issue of the underlying nodes7 library that has already been fixed. Closing the issue for now

closed 2020-03-24 by gfcittolin