st-one-io/node-red-contrib-s7 #53: Connection with Siemens MPI Adapter Error

deltafox1002, 2019-10-08:
Hello @all

My system:
Linux Mint 19 x64
SIMATIC PC Adapter USB A2 (6GK1 571-0BA00-0AA0)

I have a problem with the connection with MPI.
I installed the mpi-s7 package and also the udev rule. 

The system has found the adapter and can be found in /dev

```
root@lm19c-vm:~# ls -l /dev/mpi*
lrwxrwxrwx 1 root root 15 Okt  8 13:24 /dev/mpiusb -> bus/usb/001/003
lrwxrwxrwx 1 root root 15 Okt  8 13:24 /dev/mpiusb2 -> bus/usb/001/003
root@lm19c-vm:~#
``` 

But when i start the S7 Node i get an error message

`Error connecting to PLC: Error: Could not find any MPI-USB adater, check if the device has been correctly installed`

The user is also in group "dialout".

gfcittolin, 2019-10-08:
Hi,

could you please run `lsusb -v` as root/sudo and post the content here?

deltafox1002, 2019-10-09:
```
Bus 001 Device 003: ID 0908:01fe Siemens AG SIMATIC NET PC Adapter A2
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0908 Siemens AG
  idProduct          0x01fe SIMATIC NET PC Adapter A2
  bcdDevice            0.02
  iManufacturer           1 Siemens AG
  iProduct                2 SIMATIC PC Adapter USB A2
  iSerial                 3 SVPL7205657
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              200mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```

deltafox1002, 2019-10-09:
I think, the idProduct is different. The old version has 0004 and the new version has 01fe. I have already change the udev rule to the new id. But maybe it must be change in the mpi-s7 lib

gfcittolin, 2019-10-09:
Exactly. The mpi-s7 library searches for USB devices by their vendor and product ID. Maybe you can try manually changing it there and check if it works. My only concern is whether this newer version implements the same protocol or not

gfcittolin, 2019-10-09:
The file you'd have to change is mpi-s7/src/nodes7.js. There's a constant "MPI_USB_PID" there on line 42 that you can try putting the other product ID. Let me know the result of that, then maybe we can add as a second possible check

deltafox1002, 2019-10-10:
Hi. unfortunately that does not work. I get this error in Node-RED

`"Error connecting to PLC: TypeError: Cannot read property 'on' of undefined"`

gfcittolin, 2019-10-10:
Is there any stack trace following this message?
Maybe we can try getting more info by running node-red like the following:

    NODE_DEBUG=mpi-s7 path/to/your/node-red -v

This will make the mpi-s7 print a lot of debug info on the console. Could you eventually run your node-red instance like this and post the result?

deltafox1002, 2019-10-10:
```
[10 Oct 14:45:53 - [info] Starting flows
MPI-S7 2600: create PPPParser
MPI-S7 2600: create PPPSerializer
MPI-S7 2600: create PPPSocket
MPI-S7 2600: create MPIAdapter
10 Oct 14:45:53 - [info] Started flows
10 Oct 14:45:53 - [info] [s7 endpoint:S7-300] Initiating connection to PLC
[153,990063708] Initiate Called - Connecting to PLC with address and parameters:
[153,990129697] { mpiAddress: 2, selfMpiAddress: 0 }
[153,991282392] Connect NOW
[153,991412345] Connection cleanup is happening
[153,991516902] Clearing read PacketTimeouts
[153,991585402] Clearing write PacketTimeouts
MPI-S7 2600: new MPIAdapter
MPI-S7 2600: new PPPSocket
MPI-S7 2600: new PPPParser
MPI-S7 2600: new PPPSerializer
MPI-S7 2600: new MPIParser
MPI-S7 2600: new MPISerializer
MPI-S7 2600: MPIAdapter open
MPI-S7 2600: PPPSocket open
[153,994256791] <initiating a new connection Thu Oct 10 2019 14:45:53 GMT+0200 (CEST)>
[153,994309355] Attempting to connect to host...
[153,994721234] ConnectionReset is happening
[153,995132387] We Caught a connect error TypeError: Cannot read property 'on' of undefined
10 Oct 14:45:53 - [error] [s7 endpoint:S7-300] Error connecting to PLC: TypeError: Cannot read property 'on' of undefined
MPI-S7 2600: MPIAdapter close
MPI-S7 2600: MPIAdapter close not-connected
[155,499765730] ResetNOW is happening
[155,500238626] Clearing an earlier scheduled reset
[155,500902944] mpiAdapter closed
10 Oct 14:45:58 - [info] [s7 endpoint:S7-300] Closing connection to PLC]

```

gfcittolin, 2019-10-10:
Looks like the adapter's interface is different from all adapters we have here (and therefore the code is prepared to handle). The code expects a device configuration layout that does not match with your adapter's one while initializing the USB device, so the endpoint is left undefined, and we get this error.

Unfortunately I'd say we can't support your adapter right now, as it looks like it will require more work on the library side, and we don't have an adapter like yours here for development and testing.

deltafox1002, 2019-10-11:
Hi. I've tried it with the old version adapter but i get the same error. I roll back all changings, udev and nodes7.js to id 0004 and restart the system. 
btw, i run the linux machine in virtualbox 6. Maybe this is a problem.

gfcittolin, 2019-10-14:
I think VirtualBox shouldn't be a problem. Is the `lscpi -v` of the old one also the same? (or could you post it too?)
We expect a device with ID `ID 0908:01fe`, one interface at `0`, and, two endpoints on `0x02` and `0x82`. If it differs from it, we can try figuring out what to do to make it compatible, assuming the protocol they talk is the same.

deltafox1002, 2019-10-16:
Hi
i think you mean `lsusb -v` and not `lspci -v`

this is the output from the old version Adapter (6ES7 972-0CB20-0XA0)

**Bus 001 Device 003: ID 0908:0004 Siemens AG 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass        49 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0908 Siemens AG
  idProduct          0x0004 
  bcdDevice            0.10
  iManufacturer           1 SIEMENS AG
  iProduct                2 SIMATIC PC Adapter USB
  iSerial                 3 6ES7 972-0CB20-0XA0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          4 PC Adapter II
    bmAttributes         0xc0
      Self Powered
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         0 (Defined at Interface level)
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              5 S7
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0001
  Self Powered**

and this is the output from the new version (6GK1 571-0BA00-0AA0)

**Bus 001 Device 006: ID 0908:01fe Siemens AG SIMATIC NET PC Adapter A2
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0908 Siemens AG
  idProduct          0x01fe SIMATIC NET PC Adapter A2
  bcdDevice            0.02
  iManufacturer           1 Siemens AG
  iProduct                2 SIMATIC PC Adapter USB A2
  iSerial                 3 SVPL7205657
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              200mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)**

You write something about two endpoints. But I can see only one endpoint.

deltafox1002, 2019-10-16:
I was wrong. There are two endpoint :-)

gfcittolin, 2019-10-16:
Just for the reference, here's what the output of my ~`lspci`~ `lsusb` says:

```
Bus 003 Device 009: ID 0908:0004 Siemens AG 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass        32 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0908 Siemens AG
  idProduct          0x0004 
  bcdDevice            0.10
  iManufacturer           1 SIEMENS AG
  iProduct                2 SIMATIC PC Adapter USB
  iSerial                 3 6ES7 972-0CB20-0XA0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          4 PC Adapter II
    bmAttributes         0xc0
      Self Powered
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         0 (Defined at Interface level)
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              5 S7
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

Note that the enpoint addresses are different (`0x02, 0x82` vs `0x01, 0x82`), but everything else seems to be the same. As there's always only 2 endpoints, I can try dynamically determining the  endpoints based on their direction instead of their addresses.

As a test, would you mind changing the hardcoded addresses, so we can check if this is going to work? The file to be changed is `mpi-s7/src/ppp-usb/pppSocket.js`, on lines 18 and 19 there are the `USB_ENDPOINT_IN` and `USB_ENDPOINT_OUT` constants, you can try changing them according to what `lsusb` tells you.

deltafox1002, 2019-10-17:
I tried your suggestion but this did not work, too. There are more difference at the same adapters. The right one is yours.  

[image]

gfcittolin, 2019-10-18:
Oh nice. Have you had to change the endpoint numbers as above? What about the newer version?

deltafox1002, 2019-10-21:
Hi. The connection with the new version adapter can not be established. 

```
21 Oct 10:00:54 - [info] Flows starten
MPI-S7 1832: create PPPParser
MPI-S7 1832: create PPPSerializer
MPI-S7 1832: create PPPSocket
MPI-S7 1832: create MPIAdapter
21 Oct 10:00:54 - [info] Flows gestartet
21 Oct 10:00:54 - [info] [s7 endpoint:2ff1e2a5.2f79b6] Initialisiere Verbindung zur SPS
[290,767786173] Initiate Called - Connecting to PLC with address and parameters:
[290,767910066] { mpiAddress: 2, selfMpiAddress: 0 }
[290,769631526] Connect NOW
[290,769771501] Connection cleanup is happening
[290,769859360] Clearing read PacketTimeouts
[290,769935104] Clearing write PacketTimeouts
MPI-S7 1832: new MPIAdapter
MPI-S7 1832: new PPPSocket
MPI-S7 1832: new PPPParser
MPI-S7 1832: new PPPSerializer
MPI-S7 1832: new MPIParser
MPI-S7 1832: new MPISerializer
MPI-S7 1832: MPIAdapter open
MPI-S7 1832: PPPSocket open
MPI-S7 1832: PPPSocket _onOpen
MPI-S7 1832: PPPSerializer _transform
MPI-S7 1832: PPPSerializer serialize { seqId: 252 }
MPI-S7 1832: PPPSerializer serialize result <Buffer 7e fc 9b cd 7e>
MPI-S7 1832: PPPSocket _writeUsb <Buffer 7e fc 9b cd 7e>
[290,776301435] <initiating a new connection Mon Oct 21 2019 10:00:54 GMT+0200 (GMT+02:00)>
[290,776420189] Attempting to connect to host...
```

SEdilson, 2020-03-18:
Hi guys, i'm having the same issue with the old version. The error that i get is LIBUSB_ERROR_ACCESS

gfcittolin, 2020-03-24:
@SEdilson please check out whether you've correctly setup your environment according to the instructions on [mpi-s7](https://github.com/netsmarttech/mpi-s7)'s Readme

gfcittolin, 2020-03-24:
@deltafox1002 I'm closing this issue for now, and I'm documenting that we currently support only the "old" version of the adapter on the original "mpi-s7" repository. Thanks for your contribution

closed 2020-03-24 by gfcittolin