st-one-io/node-red-contrib-s7 #56: Problems with installing node-red

JMStoerk, 2019-11-19:
Hello everyone,

unfortunately I am a beginner with the topics node-red and iot2000.

I have a problem installing node-red.
I follow the instructions "SIMATIC_IOT2000_Setting_up_V2.1.1.pdf " at: [SIMATIC Setting up](https://support.industry.siemens.com/tf/ww/en/posts/setting-up-the-simatic-iot2000/155642/?page=0&pageSize=10)
.
During the installation I get the error message during prebuild-install  attached as a picture. For me this means as much as:
The github-Link could not be found. (If I try to open the link manually in the browser, the page cannot be found either). While searching for the node-usb package it turned out that the ia32 package for version 1.6.1,v48,linux  OS is not existing (anymore).
How can I still install node-red or replace the wrong link?

[image]

Many thanks in advance
Joachim

gfcittolin, 2019-11-20:
Hello, and welcome to GitHub.

This message is ok. It's just saying it couldn't find any pre-compiled binaries of one library we use for your operating system, and therefore it will try to compile the library locally (we can confirm that on the " CC(target) ..." line on the screenshot).

Even if this compilation step failed on your system, then just the connection to PLCs over MPI won't work; the connection over ethernet cable should work normally, regardless of this optional library failing to install.

JMStoerk, 2019-11-20:
Hello gfcittolin,

thanks for your quick answer :)
You helped me a lot.

Best regards
Joachim

closed 2020-01-06 by gfcittolin