st-one-io/node-red-contrib-s7 #58: Node says: SyntaxError: Unexpected identifier (line:38)

SammyGoderis, 2019-12-24:
[image]

The node is also not shown on the left.
IOT2000
V0.19.5

SammyGoderis, 2019-12-24:
Delete version 2.1.0 and install 2.0.2 did work.

"npm install node-red-contrib-s7@2.0.2"

gfcittolin, 2020-01-06:
You're probably using an outdated version of Node.JS. Please check by running `node -v`. The minimum supported version is 6.x

kbogusz, 2020-01-11:
After updating NodeJS to latest version I also had issues with installing s7node at all. Unfortunately I didn't save any screenshots..

gfcittolin, 2020-01-13:
Could you please try to reproduce the error and post logs and/or screenshots?

gfcittolin, 2020-03-24:
Closing the issue due to lack of activity. Please feel free to reopen it if the issue persists.

closed 2020-03-24 by gfcittolin