st-one-io/node-red-contrib-s7 #59: Catch the error with the Catch node

VictorStela, 2020-01-08:
The catch and status node don't work properly with this pallete, because the error is caused from the configuration node of the PLC connection.

I need to get the node error.

@gfcittolin, could you make this improvement?

gfcittolin, 2020-03-24:
Hi, have you tried putting the configuration on a specific flow instead of the default "On all flows"? This way you should be able to catch the errors using the Catch node without a problem.

The problem of making the individual `s7 in`, `s7 out` and `s7 control` nodes to throw errors from `s7 endpoint` is that, when multiple nodes are used with the same endpoint, all of them would throw these errors, polluting both the debug sidebar and the Node-RED logs. Do you think in a better solution for that?

gfcittolin, 2020-05-23:
Closing because of inactivity. Feel free to reopen it in the case still needed

closed 2020-05-23 by gfcittolin

jolrom, 2021-04-06:
Hello,
I got similar issue. I'm using this palette to collect data from a PLC S71500. The communication works nicely but I want to get the error when the connexion fail.
For example if I unplug my PLC I see the error in the debug sidebar (without using debug node). I want to get this error or at least the information that the connection is lost (to show it on my dashboard). I tried with the catch node but didn't succed to make it work...
I tried to "catch errors from selected nodes" with only my S7 node as mentioned on the last answer but nothing better.

[image]

Thank you in advance

OriolFM, 2024-02-09:
> Hi, have you tried putting the configuration on a specific flow instead of the default "On all flows"? This way you should be able to catch the errors using the Catch node without a problem.

I was having this issue, but when I switched the configuration node to the specific flow instead of "on all flows", the errors are caught normally by the catch node. Thanks for the help!

> The problem of making the individual `s7 in`, `s7 out` and `s7 control` nodes to throw errors from `s7 endpoint` is that, when multiple nodes are used with the same endpoint, all of them would throw [...]

As you point out, that would be chaos. I currently only use the nodes for getting data from the PLCs and not write back to them, and I use only one S7 out/control node per endpoint, so it's not a big problem for me. But if I had to also write to the PLCs, it would get complex very quickly, unless I put all the s7 nodes from one endpoint on the same tab, and do not mix them.