st-one-io/node-red-contrib-s7 #61: Unknown S7 in Object

ismetbalci, 2020-02-07:
Hello,

When I load the library on npm or on Node-red.I see the library nodes.I confıgure  the library nodes and deploy it. When I deployed the project I have an error on Node-red.That say "unknown object" and I never see again the S7 nodes.When I looked Managa pallette,the library seem installed but I am not access to library nodes.Also you can check the attached file.
Could anybody help me about the issue ?

[image]
[image]

gfcittolin, 2020-02-09:
Which version of Node.js are you using? You need at least version 8.x in able to use this node. Also try upgrading your Node-RED, from the screenshots it looks like you're using a quite old one.

gfcittolin, 2020-03-24:
Closing this issue due to lack of activity. If the problem persists even after updating Node.JS and Node-RED, please feel free to reopen it.

closed 2020-03-24 by gfcittolin