st-one-io/node-red-contrib-s7 #63: Feature: add and remove variables from Node.js

javiermrz, 2020-02-13:
Hi!

I have a Node-RED embedded instance inside Node.js. I was wondering if there is a way of adding or removing variables from the plc node from Node.js code, interacting with the RED instance. 

Or maybe there is some file where the node reads the variables from thus we can read/write into it?

I would gladly implement it if we can't do it already, if you give me some clues as where to start.

Best!

Javier

gfcittolin, 2020-02-17:
Hey, this feature is also on my personal wish list, although I haven't had the time to implement it yet. Here are my thoughts on how that should be:
 - We could add a new item/function on `s7 control' node to update the variables list from `msg.payload`. The format could be an array or an object, for example.
 - The config node's variable list would then behave as the "initial" variable list, but could be left empty, and then it would be set by the control node only
 - We should promote the part of the `onConnect()` of `S7Endpoint` that handle the variable list to a function of its own, that should be then also called by the `s7 control` node, reusing the logic.

A PR with the feature would be very much welcomed. Let me know what you think about and we can discuss it further if needed

ericcabero, 2024-06-22:
Hi,
I'm  in the same context, there is still no solutions for that?
Thank you in advance,