st-one-io/node-red-contrib-s7 #65: Input Word failing above 62 (IW62)

palmergit, 2020-03-11:
Able to load word at IW62 on S300 PLC using this node. When I try and read above this ie IW63 or IW64 it fails. I need to be able to get to IW172. Any assistance appreciated.

gfcittolin, 2020-03-24:
I've just tried it here with up to IW126 (the limit of the model I have here) without any problem. Please check the size of the memory areas you have there. If it still doesn't work, would you mind posting your flow, the part number of the PLC, and the logs (by turning the "Verbose" option on)?

palmergit, 2020-03-25:
Thank you for the reply. I will try and check on model of PLC and get the logs when I can get back into the factory. Unfortunately no access at present.

palmergit, 2020-03-26:
Ive attached Logs for IW62 (Good_Log) and IW64 (Bad_log). Also the flow. I just got a quick screenshot of log by using node-red-log. Turned on 'debug' in the settings.js Hopefully that is what you need. Will try get model.

[node red good log.txt](https://github.com/netsmarttech/node-red-contrib-s7/files/4388809/node.red.good.log.txt)
[node red bad log.txt](https://github.com/netsmarttech/node-red-contrib-s7/files/4388810/node.red.bad.log.txt)
[node red flow.txt](https://github.com/netsmarttech/node-red-contrib-s7/files/4388811/node.red.flow.txt)

gfcittolin, 2020-03-26:
From the logs, it says `Invalid Response Code - 5`. This means the PLC is complaining that the address is out of range. What is the exact Part Number (6ES7-...) of your PLC?

palmergit, 2020-03-27:
Controller is 6AU1445-2AD00-0AA0

gfcittolin, 2020-03-27:
So, this is is a SIMOTION... good to know that it even connects to it :)

This has never been tested (at least by us) against SIMOTION or SINAMICS devices, and we have zero experience with it. But taking a look at the SIMOTION catalog, I can see that they have a process image of only 64 bytes, and that would explain why you're able to access only up to IW62.
[image]

You can maybe try accessing it using the periphery image, by addressing it with `PIW64`... but no guarantee that it will work at all

palmergit, 2020-03-27:
Ah, I see! Thank you for that info. Yes it connects fine. I was treating it as 300. Will see what can do with PIW64 and report back.

palmergit, 2020-03-30:
Arrgh. Cant get it to work with now IW62 either. Would really like it to work as it would assist greatly
[node red log 300320.txt](https://github.com/netsmarttech/node-red-contrib-s7/files/4403857/node.red.log.300320.txt)
[node red flow 300320.txt](https://github.com/netsmarttech/node-red-contrib-s7/files/4403859/node.red.flow.300320.txt)

. I've attached log and single flow if you have any ideas. Thank you for your time

gfcittolin, 2020-03-30:
Good question... the PLC is replying with an error code (`0x8304`) that I've never seen before... a very quick search on google showed a possible cause of "out of resources". Maybe you're connecting/watching from multiple places, or the PLC is having not enough cycle resources for processing communication requests?

palmergit, 2020-03-31:
Where did you get that 0x8034? I'm guessing from the buffer string? What does that buffer string represent? Does it only show when there is an error? Have nothing new connected other than what I had before.

Thank you

gfcittolin, 2020-04-01:
Yes, from the buffer string. There I can identify the plc's answer type `32 02`, and then the last two bytes are the error code, should normally be `00 00`, indicating no error. This is the raw buffer as received from the PLC, and is printed only when there's something unexpected (like error code, telegram type, etc). Error handling is somewhat crude in the library we use, but it will be much better in the future.

Regarding the error itself, something must have changed... the node/library themselves shouldn't have changed, I can only think on something PLC side (or network maybe?) Unfortunately I've never seen this error, nor can I simulate it here, so I'm afraid I can't help much here :disappointed:

gfcittolin, 2020-04-01:
Just remembered this is actually a SIMOTION... so maybe it needs to be in a defined state to be able to rea

gfcittolin, 2020-04-01:
Ups, didn't meant to

gfcittolin, 2020-05-23:
Any news on that?

palmergit, 2020-05-26:
Unfortunately with lock-down at moment we haven't been able to get back into factory to try anything new. We are hoping to be able to try again in the next couple of weeks.

palmergit, 2020-06-09:
Thank you for your help. We are now able to read data from the PLC again and are trying a different area to get our data from. Now having and issue trying to read LREAL data from a DB. Not sure how to go about this. You can close this original case though. Thank you.

closed 2020-11-03 by GlobalAutomation