st-one-io/node-red-contrib-s7 #67: Variable addressing

connelltd, 2020-03-19:
Playing around with this.. had some success.. 

Using an S7-200 I'm able to connect to the PLC, able to control/read the digital I/O using I0.0/Q0.0 for example.. but struggling a little to understand correct addressing or format for an analog in/out or any EM (expansion module) for that matter.

Currently I have plugged in an EM AQ04 analog output module, configured in Step 7, and can send a 16bit integer to AQW16 which outputs 0~10V, but I try sending a number to the same variable and nothing happens, no debug or errors shown.

I suspect it is a format issue on my end so looking for some guidance

gfcittolin, 2020-03-24:
I don't have much experience with S7-200, nor I have one here for testing. But some people have already done this with this node. Just make sure you use the correct variable format, as shown in the info panel of the configuration node, or in the Readme of this repo.

As far as I know, to talk to S7-200 you need the same variable scheme as used on a LOGO! PLC. Please check https://www.promotic.eu/en/pmdoc/Subsystems/Comm/PLC/LOGO.htm where they show a "conversion table". In your case, `AQ16` would translate to `DB1,W1102` if I'm not doing anything wrong

connelltd, 2020-03-25:
Appreciate the help, trying that address didn't seem to work, but i'll try going through the range to see if I can find it.

I have a wireshark capture of port activity on my PC when manually changing the analog output to 7777 when using STEP-7, I'm sure to someone the correct address would be clear as day :)
   
[image]

As a side note, when using the variable format `DB1,W1102` the variable payload is ok to be in 'number' format or should it use 'string' or does it not matter? 

Reason is for test purpose I can set `Q0.0 = true` and that works to set output on PLC, if I read `Q0.0` it returns 1, again good. But if I try reading `DB1,X1064.0` I get 0, and same if I try `DB1,W1064` in an attempt to read word of the entire output pins.

gfcittolin, 2020-03-26:
Yes, it's now clear as day :) The issue is the "Area" field of the item. In your wireshark dump, it has the value of `0x07`, but this area is supported neither by the library we use, consequently nor by us. We currently support the areas P, I, Q, M, DB (`0x80` to `0x84`), and C and T (`0x1c` and `0x1d`).

For now, I think the easiest way to solve this is to read/write to the M area, and then manually copy it to the desired analog output from the PLC program.

That said, there's a rewrite of the library undergoing. We don't have any S7-200 here so that we could implement it and fully test it, but if you could provide a wireshark dump of you reading and writing, single and multiple analog inputs and outputs, we could prepare it already to support these area. Do you think it's possible?

gfcittolin, 2020-03-26:
By the way, It looks like I was completely wrong regarding using `DB1` as an addressing method. I think it's actually only valid for the LOGO! series.

> As a side note, when using the variable format DB1,W1102 the variable payload is ok to be in 'number' format or should it use 'string' or does it not matter?

If I got your question right, the "W" here stands for "Word" that, in the PLC world, means a 16-bit (2-byte) long address area holding and unsigned number (that is, in the range 0~65535), so no relation to being a number or a string. Despite of that, if you supply a string with just a number inside it (e.g. `"1234"`), it will be automatically casted to a number before writing it to the PLC.

connelltd, 2020-03-30:
> if you could provide a wireshark dump of you reading and writing, single and multiple analog inputs and outputs, we could prepare it already to support these area. Do you think it's possible?

Yup no problem, I have access to a number of EM modules so can read/write to all if would help (AE04, AR02, AR04, AQ04, AT04).