st-one-io/node-red-contrib-s7 #69: Error: Could not load MPI-USB library, check installation

Waterboy1602, 2020-03-28:
I try to connect my PLC S7-200 with a PPI cable to Node-RED. Every time I give in the properties for an S7 endpoint with "Transport" on "MPI/PPI/Dp USB Adapter", I get this error: "Could not load MPI-USB library, check your installation.

I already tried a couple of times to reinstall it, but I get all the time the same error. How can I fix this?

gfcittolin, 2020-04-01:
We currently only support MPI/PPI/DP adapters [like this](https://support.industry.siemens.com/cs/pd/10896?pdti=pi&lc=en-ww), so using an USB/Serial converter and connecting using a serial cable is not supported by this node

Waterboy1602, 2020-04-01:
Okay, thank you for the clarification! I ordered such a cable, hopefully I will get it to work with the other cable.

Waterboy1602, 2020-04-19:
@gfcittolin I received my cable. I installed everything correctly with Node-Red, contrib-s7 and mpi-s7. I also copied the udev 90 file to the right place. But when I now try to start node-red I get this error `LIBUSB_ERROR_ACCESS`. If I try to start node-red with `sudo nod-red`, I get this error `Waiting for missing types to be registered: s7 endpoint, s7 in, s7 out`

How can I fix this error?

gfcittolin, 2020-04-20:
So, the two different errors have two different causes.

The `LIBUSB_ERROR_ACCESS` is due to lack of permissions. The udev rule just makes sure that the users part of the `dialout` group have access to the adapter. You still need to make sure that the user running Node-RED (that is, your current user, if you're running from the terminal) is part of the `dialout` group

Running node-red as root (although not recommended as a final solution) should make the device accessible. The most probable cause you're seeing the `missing types` error due to the fact that, when running Node-RED with `sudo`, you're running it from another user (the `root` user), so the paths for the userDir of Node-RED (including flows and locally installed nodes) are different, and need to be ajusted.

I suggest you _not_  to run Node-RED as root, and instead fix the permissions of your system

Waterboy1602, 2020-04-20:
I've succeeded in adding my user to the `dialout` group. But I'm still getting errors. In node-red I get these two errors: `Connection step 1 refused with command [128] != [208] CONNECTION_RESPONSE` and `LIBUSB_TRANSFER_NO_DEVICE`

What could be the reason for these errors and how can I fix them?

gfcittolin, 2020-04-22:
`LIBUSB_TRANSFER_NO_DEVICE` usually happens when the device is disconnected, either physically or logically. You can confirm this by checking the kernel logs for device disconnection. Among the causes I would suspect, you can check the following:
 - Poor USB cable, or cable too long, or even whether you're getting interference from nearby sources (like big motors, frequency inverters, etc.)
 - Bad terminator resistor configuration on the bus side
 - Bus address conflict, be sure nothing is using bus address 0.

There's not much documentation on the adapter's protocol, not its error codes (as far as I know, if you find something let me know), therefore I can't tell you a precise cause of the issue. I'd start from the list above, and let me know about the results

Waterboy1602, 2020-04-22:
Thanks for the fast response and explanation!

So I tried a couple of things, all without success. I first tried to use different cables with my adapter. They all gave the same errors. The cables I normally use are the ones from the box (short: not more then 1m). [Maybe a fact that I should add. I bought a Chinese 'fake' adapter: 6ES7972-0CB20-0XA0]

Then I tried to catch some extra information from within my setup. I use an Ubuntu VirtualBox with node-red installed on it. It was clearly from `dmesg` that my adapter constantly disconnects. But I can't quite catch why exactly. Mostly just because I don't get a lot from what is going on. It would be wonderful if you could help me.

On the next links you can read all my collected logs:

- [dmesg](https://pastebin.com/JdcghjGR)
- [NODE_DEBUG=mpi-s7 node-red -v](https://pastebin.com/NjTphKZS)
- lsusb
```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 007: ID 0908:0004 Siemens AG SIMATIC PC Adapter USB
Bus 002 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```
- lsusb -v
```
Bus 002 Device 006: ID 0908:0004 Siemens AG SIMATIC PC Adapter USB
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass        32 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0908 Siemens AG
  idProduct          0x0004 
  bcdDevice            0.10
  iManufacturer           1 SIEMENS AG
  iProduct                2 SIMATIC PC Adapter USB
  iSerial                 3 6ES7 972-0CB20-0XA0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          4 PC Adapter II
    bmAttributes         0xc0
      Self Powered
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         0 
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              5 S7
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

gfcittolin, 2020-05-23:
Sorry for the delay, been pretty busy lately. Thanks for the logs. I took a look on it, and looks we're sending something at the communication setup part (before sending any S7 Communication packet) that either the PLC or the adapter is not very happy about. We send a connection request, and we get an answer with a disconnection request. Not sure why the adapter is "crashing" though.

Unfortunately, there's no known documentation about the protocol, and what have implemented so far is based on tests with the PLCs we have. In other words, I'm not sure where exactly the problem is, although I have some ideas.

If you have the PLC's project, micro/win for programming, and some time available for helping, we could try watching the communication and checking what is different from what we're sending, and hopefully figure out what has to be fixed. Let me know, and I can prepare some instructions on how to do it!

Waterboy1602, 2020-05-23:
Don't worry about it. It are hard times for everybody.

I would love it  to help you in any possible way. I can use my cable with the software that works with my S7-200 PLC: Step7 MicroWin v4. 

So let me known what I can do to help this project and I will do it as fast as possible. I would like to make some time free for this project. I think I have to say that I'm not so experienced with Siemens PLC and MicroWin, but I can do the basics and I'm always eager to learn new things.

gfcittolin, 2020-05-24:
Thanks for the availability! What we basically need is to capture the USB traffic between MicroWin and the USB adapter using Wireshark, specially the connection part. So, basically, we need the following:
 1. Install and setup [Wireshark](https://www.wireshark.org/download.html) in the same machine where running MicroWin
 2. Start capturing on the USBPcap interface
 3. Connect the USB Adapter to the PC
 4. Start MicroWin, go online, do some stuff (like monitor the running code, diagnostics, etc.), then go offline.
 5. Stop packet capture and save the resulting file

In the case you're able and allowed to change connection parameters (like the bus address of the PLC or MicroWin, connection speed, etc), it would be even better to do so while recording on Wireshark (and taking notes on what is being changed), so we can check which parameters are changing. The configs you can change can be found on the screenshots of the [S7-200 Programming Manual](https://support.industry.siemens.com/cs/document/1109582/s7-200-system-manual?dti=0&lc=en-FR), section 7.

Note that the capture will include all your USB traffic, which may include mouse and keyboard (if they're USB, of course) or other sensible traffic. So please e-mail it directly to me, for security reasons. If the file is too big, let me know and I can provide you a secure drop folder for it.

Let me know about the results or if you have any issue in your setup!

Waterboy1602, 2020-05-24:
I'm done with recording the logs with Wireshark and MicroWin. I will now send them to you by mail. 

Let me know if you need some more logs or if I need to change some extra settings.

Waterboy1602, 2020-06-08:
@gfcittolin Is there already any progress with the development about my version of a MPI-cable?

sniicker, 2020-06-18:
> 
> 
> I installed everything correctly with Node-Red, contrib-s7 and mpi-s7. I also copied the udev 90 file to the right place. 
> 
> 

@Waterboy1602  Can you pls tell us what you did here? Cannot find mpi-S7 in node red? What file did you copie? Pls help
thx sniicker

Waterboy1602, 2020-06-18:
> > I installed everything correctly with Node-Red, contrib-s7 and mpi-s7. I also copied the udev 90 file to the right place.
> 
> @Waterboy1602 Can you pls tell us what you did here? Cannot find mpi-S7 in node red? What file did you copie? Pls help
> thx sniicker
I installed this repo and the repo: [MPI-s7](https://github.com/netsmarttech/mpi-s7). There you can find the steps you need to perform to install the mpi-S7 node.

Don't forget to copy the udev rule. It is also explained in the readme.md file

gfcittolin, 2020-06-18:
Sorry, I couldn't test anything yet, since we're changing physical installations here and I don't have the PLCs available. Hope to test something within the next two weeks

Waterboy1602, 2020-07-30:
Is there already any progress or anything I can help you with?

gfcittolin, 2020-08-27:
Sorry for the delayed response, there have been some progress regarding this issue. 

First of all, there was some error handling missing in the connection phase that I've now implemented, and would prevent it from working properly. Secondly, the bus parameters when connecting to S7-200 are indeed different, and the code was not prepared for that. There are still some parameters I couldn't figure out its meaning, but at least we now know what is different and what we should send.

Last week I finally got access to a S7-200 nearby here (the factory had access limitations due to covid), and I could test these changes by manually crafting the bus parameters, and I could establish the communication and exchange variables successfully.

The missing piece of work that needs to be done (and that's why this is still not commited) is to create an interface in the code for these parameters. Still need to figure out whether we'll be able to automatically negotiate these, or whether we'll need to introduce new configuration parameters on the config node to allow the selection of them. This is somewhat critical, because we can't freely add and remove configuration parameters to the flow without the risk of breaking somebody's flow in the future.

I'm unfortunately somewhat busy developing multiple things at the same time, but as soon as I'm able to, I'll try to create a branch with these changes, so you could manually install it and test whether it works in your setup.

Waterboy1602, 2020-09-23:
Thank you for the update! 

Will you keep me up to date when the new branch is available, so that I can directly start testing it?

gfcittolin, 2020-10-20:
I've finally been able to work on this. It was a huge amount of work, not only implementing the needed bits, but also restructuring lots of things so that we can keep building and improving it.

The project is now split in two: `node-red-contrib-s7` and `node-red-contrib-mpi-s7`. The original `node-red-contrib-s7` project now handles only ethernet connections natively, but can now plug onto other transports. That's why there's the new `node-red-contrib-mpi-s7`. So now, people that only need ISO-on-TCP don't need to install and load the MPI and USB dependencies.

Therefore, for connecting to a PLC over MPI/DP or PPI, one needs to install **both** `node-red-contrib-s7` and `node-red-contrib-mpi-s7`.

They are not yet published for general availability, but already available on side channels. @Waterboy1602 if you want to try them, please run:

    npm install node-red-contrib-s7@canary

and

    npm install st-one-io/node-red-contrib-mpi-s7

Looking forward to your feedback and (hopefully not) your stack traces!

Waterboy1602, 2020-10-22:
I tried to install the two nodes, but I get some problems I can't place.

So I installed the nodes with the commands you gave me. I executed them in the folder `~/.node-red`. Then I start Node-RED, but can't find the s7-endpoint node. So I went looking in Manage palette and saw this: 
[image]

When starting Node-RED I get the follow output:
```
22 Oct 16:25:17 - [info] Node-RED version: v1.2.2
22 Oct 16:25:17 - [info] Node.js  version: v12.19.0
22 Oct 16:25:17 - [info] Linux 5.4.0-52-generic x64 LE
22 Oct 16:25:17 - [info] Loading palette nodes
22 Oct 16:25:18 - [warn] ------------------------------------------------------
22 Oct 16:25:18 - [warn] [node-red-contrib-mpi-s7/mpi-s7] TypeError: usb.unrefHotplugEvents is not a function
22 Oct 16:25:18 - [warn] [node-red-contrib-s7/s7] SyntaxError: Unexpected token '.'
22 Oct 16:25:18 - [warn] ------------------------------------------------------
22 Oct 16:25:18 - [info] Settings file  : /home/wannes/.node-red/settings.js
22 Oct 16:25:18 - [info] Context store  : 'default' [module=memory]
22 Oct 16:25:18 - [info] User directory : /home/wannes/.node-red
22 Oct 16:25:18 - [warn] Projects disabled : editorTheme.projects.enabled=false
22 Oct 16:25:18 - [info] Flows file     : /home/wannes/.node-red/flows_NodeRed.json
22 Oct 16:25:18 - [info] Creating new flow file
```

gfcittolin, 2020-10-23:
Hey,

for the `node-red-contrib-mpi-s7` issue, the problem was actually fixed on the library and a new version release, but `package-lock.json` had pinned the old version. I've just released a new version with the updated package-lock and it should work now.

for the second one, i was using node.js features only available in node 14.x. I've replaced them and it should now at least load the nodes properly

Waterboy1602, 2020-10-25:
After removing `node-red-contrib-mpi-s7` and reinstalling it the error stays: `[node-red-contrib-mpi-s7/mpi-s7] TypeError: usb.unrefHotplugEvents is not a function`. Could it be there is an other reason this error appears?

This error `[node-red-contrib-s7/s7] SyntaxError: Unexpected token '.'` disappeared. So that's fixed. Thanks for that!

gfcittolin, 2020-10-26:
> Could it be there is an other reason this error appears?

Not really. This error is about a new function I had to implement into the USB library, and all the code should install and depend on this fork. I just double-checked all `package.json` and `package-lock.json`s involved, and all seem correct. I have also installed on a separate device, and it also worked. I may have still missed something, but I'm starting to think it may be something at your side.

Could you please run `npm ls` in your Node-RED user directory (`/home/wannes/.node-red`) *and* in your Node-RED user directory, and post its contents here? If you prefer, you can also send them by e-mail.

Waterboy1602, 2020-10-26:
```
wannes@NodeRed:~/.node-red$ npm ls
node-red-project@0.0.1 /home/wannes/.node-red
├─┬ node-red-contrib-mpi-s7@1.0.0-beta.1
│ └─┬ mpi-s7@0.3.1
│   └─┬ usb@1.6.3 (github:gfcittolin/node-usb#b140b1cc9343523e6e250afde7753c9a4864d117)
│     ├─┬ bindings@1.5.0
│     │ └── file-uri-to-path@1.0.0
│     ├── nan@2.13.2
│     └─┬ prebuild-install@5.3.6
│       ├── detect-libc@1.0.3
│       ├── expand-template@2.0.3
│       ├── github-from-package@0.0.0
│       ├── minimist@1.2.5
│       ├── mkdirp-classic@0.5.3
│       ├── napi-build-utils@1.0.2
│       ├─┬ node-abi@2.19.1
│       │ └── semver@5.7.1
│       ├── noop-logger@0.1.1
│       ├─┬ npmlog@4.1.2
│       │ ├─┬ are-we-there-yet@1.1.5
│       │ │ ├── delegates@1.0.0
│       │ │ └─┬ readable-stream@2.3.7
│       │ │   ├── core-util-is@1.0.2
│       │ │   ├── inherits@2.0.4 deduped
│       │ │   ├── isarray@1.0.0
│       │ │   ├── process-nextick-args@2.0.1
│       │ │   ├── safe-buffer@5.1.2 deduped
│       │ │   ├─┬ string_decoder@1.1.1
│       │ │   │ └── safe-buffer@5.1.2 deduped
│       │ │   └── util-deprecate@1.0.2
│       │ ├── console-control-strings@1.1.0
│       │ ├─┬ gauge@2.7.4
│       │ │ ├── aproba@1.2.0
│       │ │ ├── console-control-strings@1.1.0 deduped
│       │ │ ├── has-unicode@2.0.1
│       │ │ ├── object-assign@4.1.1
│       │ │ ├── signal-exit@3.0.3
│       │ │ ├─┬ string-width@1.0.2
│       │ │ │ ├── code-point-at@1.1.0
│       │ │ │ ├─┬ is-fullwidth-code-point@1.0.0
│       │ │ │ │ └── number-is-nan@1.0.1
│       │ │ │ └── strip-ansi@3.0.1 deduped
│       │ │ ├─┬ strip-ansi@3.0.1
│       │ │ │ └── ansi-regex@2.1.1
│       │ │ └─┬ wide-align@1.1.3
│       │ │   └── string-width@1.0.2 deduped
│       │ └── set-blocking@2.0.0
│       ├─┬ pump@3.0.0
│       │ ├─┬ end-of-stream@1.4.4
│       │ │ └── once@1.4.0 deduped
│       │ └─┬ once@1.4.0
│       │   └── wrappy@1.0.2
│       ├─┬ rc@1.2.8
│       │ ├── deep-extend@0.6.0
│       │ ├── ini@1.3.5
│       │ ├── minimist@1.2.5 deduped
│       │ └── strip-json-comments@2.0.1
│       ├─┬ simple-get@3.1.0
│       │ ├─┬ decompress-response@4.2.1
│       │ │ └── mimic-response@2.1.0
│       │ ├── once@1.4.0 deduped
│       │ └── simple-concat@1.0.1
│       ├─┬ tar-fs@2.1.0
│       │ ├── chownr@1.1.4
│       │ ├── mkdirp-classic@0.5.3 deduped
│       │ ├── pump@3.0.0 deduped
│       │ └─┬ tar-stream@2.1.4
│       │   ├─┬ bl@4.0.3
│       │   │ ├─┬ buffer@5.6.1
│       │   │ │ ├── base64-js@1.3.1
│       │   │ │ └── ieee754@1.1.13
│       │   │ ├── inherits@2.0.4 deduped
│       │   │ └─┬ readable-stream@3.6.0
│       │   │   ├── inherits@2.0.4 deduped
│       │   │   ├── string_decoder@1.1.1 deduped
│       │   │   └── util-deprecate@1.0.2 deduped
│       │   ├── end-of-stream@1.4.4 deduped
│       │   ├── fs-constants@1.0.0
│       │   ├── inherits@2.0.4
│       │   └─┬ readable-stream@3.6.0
│       │     ├── inherits@2.0.4 deduped
│       │     ├── string_decoder@1.1.1 deduped
│       │     └── util-deprecate@1.0.2 deduped
│       ├─┬ tunnel-agent@0.6.0
│       │ └── safe-buffer@5.1.2
│       └── which-pm-runs@1.0.0
└─┬ node-red-contrib-s7@3.0.0-beta.3
  └─┬ @st-one-io/nodes7@1.0.0-beta.1
    └── iso-on-tcp@0.1.0
```

gfcittolin, 2020-10-27:
Just saw there's a typo in my previous comment, i meant to get the `npm ls` from your Node-RED user directory and your Node-RED **installation** directory. My guess was that, if it has been installed together with Node-RED itself, it would be loading that copy instead of the one in the user directory. The structure of modules in your userdir seems correct, though.

Could you check if there's any copy of the `usb` node or the `mpi-s7` node in your Node-RED's installation directory (or post `npm ls` here)? Additionally, could you run the following into inside `~/.node-red` and check the output?

    node -p "require('usb').unrefHotplugEvents"

The correct output is `[Function: unrefHotplugEvents]`, and if you get it, then the installation is indeed correct and the problem must be somewhere else in the environment

Waterboy1602, 2020-10-27:
This is the `npm ls` from my installation directory: [https://pastebin.com/ZEXHhPpc](https://pastebin.com/ZEXHhPpc)

After that I tried to execute `node -p "require('usb').unrefHotplugEvents"` in `/home/wannes/.node-red`, but got this output `undefined` instead of the correct output.

Waterboy1602, 2020-10-29:
I've found something else. When I try to install `npm install node-red-contrib-mpi-s7` with nodejs V14.15.0 en npm V6.14.8 I get the following error output. Could it be this has something to do with the reason I dont get to work it with nodejs V12.X?

```
> usb@1.6.3 install /home/wannes/.node-red/node_modules/usb
> prebuild-install --verbose || node-gyp rebuild

prebuild-install info begin Prebuild-install version 5.3.6
prebuild-install info looking for cached prebuild @ /home/wannes/.npm/_prebuilds/91abd0-usb-v1.6.3-node-v83-linux-x64.tar.gz
prebuild-install http request GET https://github.com/tessel/node-usb/releases/download/v1.6.3/usb-v1.6.3-node-v83-linux-x64.tar.gz
prebuild-install http 404 https://github.com/tessel/node-usb/releases/download/v1.6.3/usb-v1.6.3-node-v83-linux-x64.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=14.15.0 runtime=node arch=x64 libc= platform=linux)
make: Entering directory '/home/wannes/.node-red/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
../libusb/libusb/os/linux_udev.c:40:10: fatal error: libudev.h: No such file or directory
   40 | #include <libudev.h>
      |          ^~~~~~~~~~~
compilation terminated.
make: *** [libusb.target.mk:150: Release/obj.target/libusb/libusb/libusb/os/linux_udev.o] Error 1
make: Leaving directory '/home/wannes/.node-red/node_modules/usb/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/wannes/.nvm/versions/node/v14.15.0/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)
gyp ERR! System Linux 5.4.0-52-generic
gyp ERR! command "/home/wannes/.nvm/versions/node/v14.15.0/bin/node" "/home/wannes/.nvm/versions/node/v14.15.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/wannes/.node-red/node_modules/usb
gyp ERR! node -v v14.15.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! usb@1.6.3 install: `prebuild-install --verbose || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the usb@1.6.3 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```

gfcittolin, 2020-10-30:
Oh, I think you may have found it. We need `libusb` in order to have raw access the USB devices, itself needing `libudev` to work. Installing `mpi-s7` should trigger compiling and installing everything, and it looks like you don't have the needed dependencies for that to happen. What I still can't understand is, given that you don't have the dependencies to compile, how the installation with the older version of node succeeded in the first place? :thinking: 

Please try to install `libudev-dev` (that provides the missing header above) and `build-essential` (if on debian-based system, with `sudo apt install build-essential libudev-dev`) and try again.

gfcittolin, 2020-10-30:
Ok, found it. There's this package in the original usb repository called `prebuild-install`, that checks the repository for prebuilt binaries, and install them if it finds one compatible with the running platform, which was the case for your node 12.

According to [this](https://github.com/prebuild/prebuild-install#help), you may be able to install it correctly by passing `--build-from-source` on npm's command line. You may need to uninstall the packages first before trying to install it with this flag.

Waterboy1602, 2020-11-02:
I succeeded in installing `node-red-contrib-mpi-s7` by first installing `libudev-dev` and `build-essential`. After that I executed the npm command with `--build-from-source` and got no errors anymore.

But now if I connect my MPI adapter and fully configure it, node-red gives me the following error: `Error: Connection step 1 refused with command [128] != [208] CONNECTION_RESPONSE`. I use the adapter with a short cable, so that can't be the problem. I also checked `lsusb` and this is my output:
```
Bus 001 Device 004: ID 0908:0004 Siemens AG SIMATIC PC Adapter USB
Bus 001 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

gfcittolin, 2020-11-03:
This is already much better. At the point of this error, we're already connected to the adapter, bus is configured, and we're trying to connect to the PLC directly, but it's refusing our connection request.

Comparing the dump of my tests with an S7-200 with the one you've sent me, there's indeed a difference in the connection command, but I don't know exactly why that is different. If you could send me the project of your PLC by e-mail, I can try to figure out what that means.

In the meantime, in order to confirm that, you can try to manually change that value and then check if it works. Please change the file `~/.node-red/node_modules/mpi-s7/src/mpi-adapter/mpiSerializer.js`, and on **line 212** replace the value `0` of

                        buf.writeUInt8(0, 17); //rack+slot

with the value `2`:

                        buf.writeUInt8(2, 17); //rack+slot

and restart Node-RED. Please then activate the logging of the library when running Node-RED by starting it like the following (adjusting eventually for your environment):

    NODE_DEBUG=mpi-s7 node-red

There will be a lot of logs, and it would be awesome if you could post them after the change.

Thank you for your help and your patience!

Waterboy1602, 2020-11-03:
Thank you for all the effort you're still putting into this!

I've changed value `0` to `2` on line 212 of `mpiSerializer.js` and started Node-Red with the extra variables. This was the output I got: [https://pastebin.com/snxYDtK4](https://pastebin.com/snxYDtK4).

I've also send my PLC project by mail to you.

Waterboy1602, 2020-12-03:
Some things I could help you with while debugging this problem?

Waterboy1602, 2021-11-24:
I stopped trying to let the communication work with S7-200 by MPI. I instead got my hands on a Ewon Ebuddy module that let me communicate over ethernet.