st-one-io/node-red-contrib-s7 #72: connection using MPI

binmulyadin, 2020-06-16:
Firstly, thanks for your awesome jobs. I am sorry if this is not the right place to ask this question.
I  to get communication to Simatic S7-400 without any problem by using ethernet.

I wonder how to get connection using MPI Communication since some PLC doesn't have Ethernet connection. I am using USB-MPI PC Adapter, I also has USB A2 PC Adapter.

Please advice
regards
bin mulyadin

niox1, 2020-06-16:
I would also be interested in the same thing.
I use NodeRed on a Windows computer. The PC adapter is installed. The PLC can be reached with TIA.
I get the error message: "Error connecting to PLC: Error: LIBUSB_ERROR_NOT_SUPPORTED"

regrads
niox

sniicker, 2020-06-18:
I'm on the same Step as @niox1 
Connection is working with Step7 and TIA V15.1
But get the same error: **LIBUSB_ERROR_NOT_SUPPORTED**

Readme says:  _An optional compilation step is only necessary for supporting MPI-USB adapter._

What steps? And where can we find this informations?

Pls help!
THX sniicker

sniicker, 2020-06-18:
The Error:  **LIBUSB_ERROR_NOT_SUPPORTED** means that you have not installed the WINusb driver.
Whatch here: 
[https://github.com/tessel/node-usb](url)
[https://github.com/netsmarttech/mpi-s7](url)

I installed the Zadig Software and the libusb-win32 driver.

Now USB Adapter seems to work, but i get another error:  **Error connecting to PLC : TypeError: Cannot read property 'on' of undefined**

What to do now??

regards sniicker

gfcittolin, 2020-06-18:
Hi all,
Support for MPI/DP is still somewhat experimental, and is not supported on Windows for now, as this is only being tested on Linux. 

@binmulyadin 
as said, support for MPI/DP is still experimental. The A2 version of the adapter is not supported by this node, but if you have an old one and want to experiment, you can follow the instructions posted above by sniicker

@sniicker 
Your error doesn't seem to be related to USB at a first glance, do you have a full stack trace of it?

binmulyadin, 2020-06-19:
> Hi all,
> Support for MPI/DP is still somewhat experimental, and is not supported on Windows for now, as this is only being tested on Linux.
>
> @binmulyadin
> as said, support for MPI/DP is still [...]

understood, then I should close this issue. right?

closed 2020-06-19 by binmulyadin