st-one-io/node-red-contrib-s7 #77: Error connecting to PLC: [object Object]

next36eee, 2020-08-14:
### Describe the bug
This library works like a charm, i found a bug that even after deleting the S7 nodes from my flow but it is continuously giving an error message in the Debug which says that Error connecting to PLC: [object Object] 

### Environment
 - **Version** of this node: 2.2.1
 - **Node-RED version**: 1.1.3
 - **OS**: Windows 10
 - **Browser**: chrome

### Additional context
Add any other context about the problem here.

closed 2020-08-17 by next36eee

sniicker, 2021-02-24:
What was your Solution - same problem here

next36eee, 2021-02-24:
@sniicker In the configuration node(can be found in the options menu), you have to delete the S7-end point and then you will not get any error

sniicker, 2021-02-24:
I cannot delete the node, this is my active s7 endpoint