st-one-io/node-red-contrib-s7 #78: Can't install on Raspberry Pi

eliasz666, 2020-09-10:
After install Node-Red with this:
`bash <(curl -sL https://raw.githubusercontent.com/node-red/linux-installers/master/deb/update-nodejs-and-nodered)`

i'm always getting this:

```
pi@raspberrypi:~ $ npm install node-red-contrib-s7

> usb@1.6.3 install /home/pi/node_modules/usb
> prebuild-install --verbose || node-gyp rebuild

prebuild-install info begin Prebuild-install version 5.3.5
prebuild-install info looking for cached prebuild @ /home/pi/.npm/_prebuilds/470134-usb-v1.6.3-node-v72-linux-arm.tar.gz
prebuild-install http request GET https://github.com/tessel/node-usb/releases/download/v1.6.3/usb-v1.6.3-node-v72-linux-arm.tar.gz
prebuild-install http 404 https://github.com/tessel/node-usb/releases/download/v1.6.3/usb-v1.6.3-node-v72-linux-arm.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=12.18.3 runtime=node arch=arm libc= platform=linux)
make: Wejście do katalogu '/home/pi/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
../libusb/libusb/os/linux_udev.c:40:10: fatal error: libudev.h: Nie ma takiego pliku ani katalogu
 #include <libudev.h>
          ^~~~~~~~~~~
compilation terminated.
make: *** [libusb.target.mk:148: Release/obj.target/libusb/libusb/libusb/os/linux_udev.o] Błąd 1
make: Opuszczenie katalogu '/home/pi/node_modules/usb/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Linux 5.4.51-v7l+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/node_modules/usb
gyp ERR! node -v v12.18.3
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.6.3 (node_modules/usb):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.6.3 install: `prebuild-install --verbose || node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

+ node-red-contrib-s7@2.2.1
updated 1 package and audited 68 packages in 54.731s

2 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities

pi@raspberrypi:~ $
```

I have newest version of raspibian on two devices as Raspberry Pi 4 arm7. Please help. I need this on my project now :(

eliasz666, 2020-09-11:
Problem with installation was fixed by installation on node-red system (Manage palletes -> Install) but still is impossible by Raspberry Pi LXTerminal.

gfcittolin, 2020-10-20:
The log above is not a fatal error. It is an optional dependency that failed to install, but is not needed for normal operation of the node. If you look close at the latest log lines:

```
....

+ node-red-contrib-s7@2.2.1
updated 1 package and audited 68 packages in 54.731s
```

... you'll see that it was indeed installed.

Don't worry, there will be a new major release soon that will remove this dependency. For now, just ignore it. I'm closing this issue as there's nothing to change or to fix for now.

closed 2020-10-20 by gfcittolin