st-one-io/node-red-contrib-s7 #80: Add a Italian translate for node-red-contrib-s7

Ivan0x62, 2020-10-27:
I have translated your library into Italian, can you add the translation to the original repository?

thank you very much for your work
the library works great!

Tested with Tia portal V14 , node-red 1.0.3

repository: node-red-contrib-s7-Italian-translation
directory: node-red-contrib-s7-Italian-translation/red/locales/it-IT/s7.json

Best regards

gfcittolin, 2020-12-09:
Hi there! Could you please open a PR for that against master? This way we can keep your user name as a contributor of the project!

Ivan0x17, 2021-12-09:
> Hi there! Could you please open a PR for that against master? This way we can keep your user name as a contributor of the project!

How can I open a pull request I have never delved into

fernandoamorim1703, 2022-06-23:
We will be closing this issue, if there is a need to revisit the subject, please open a new one. Thank you very much.

closed 2022-06-23 by fernandoamorim1703