st-one-io/node-red-contrib-s7 #81: error message lacking information

SGStino, 2020-10-30:
### Describe the bug
``` node.error(RED._("s7.error.onconnect") + err.toString(), {});```

`err` is an object, so the message is appended with [object Object].
I'd use JSON to have more information about the error.

### To Reproduce
Steps to reproduce the behavior:
Manage to get an object error in onConnect

### Expected behavior
A more verbose error message

 

### Logs
```30 Oct 15:34:28 - [error] [s7 endpoint:Balenpers] Error connecting to PLC: [object Object]```

### Flow
```[{"id":"5f5824a6.20a30c","type":"tab","label":"Flow 1","disabled":false,"info":""},{"id":"6e479ba9.f799e4","type":"s7 in","z":"5f5824a6.20a30c","endpoint":"11366239.90b83e","mode":"all-split","variable":"stap","diff":true,"name":"","x":380,"y":260,"wires":[["7b3b2c13.9351f4"]]},{"id":"7b3b2c13.9351f4","type":"debug","z":"5f5824a6.20a30c","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":630,"y":260,"wires":[]},{"id":"11366239.90b83e","type":"s7 endpoint","z":"","transport":"iso-on-tcp","address":"192.168.48.252","port":"102","rack":"0","slot":"3","localtsaphi":"01","localtsaplo":"00","remotetsaphi":"01","remotetsaplo":"00","connmode":"rack-slot","adapterauto":true,"adapterport":"","busaddr":"2","adapteraddr":"0","cycletime":"500","timeout":"1500","verbose":"default","name":"","vartable":[{"addr":"DB141,WORD8","name":"stap"}]}]```

gfcittolin, 2020-11-30:
Please try the new 3.0.0 branch, among many improvements, we have new error handling code there that solves this annoying issue

gfcittolin, 2020-12-09:
Closing for now, please feel free to reopen if this still persists after upgrading to 3.0.0

closed 2023-07-28 by fernandoamorim1703