st-one-io/node-red-contrib-s7 #82: V3.0.0-beta.3 not working on IOT2040

bergmanu, 2020-11-10:
### Describe the bug
With version 3.0.0-beta.3 node-red crashes immediately when a PLC (S7-1500 / S7-1200) is connected.
The message is: 
```
9 Nov 22:18:06 - TypeError: stream.off is not a function
at ISOOnTCPClient.isoOnTcp.createConnection (/usr/lib/node_modules/node-red-                                                                                                             contrib-s7/node_modules/@st-one-io/nodes7/src/s7endpoint.js:207:24)
at Object.onceWrapper (events.js:313:30)
at emitNone (events.js:106:13)
at ISOOnTCPClient.emit (events.js:208:7)
at process.nextTick (/usr/lib/node_modules/node-red-contrib-s7/node_modules/                                                                                                             iso-on-tcp/src/client.js:193:45)
at _combinedTickCallback (internal/process/next_tick.js:132:7)
```

### To Reproduce
Steps to reproduce the behavior:
1. Install node-red-contrib-s7 via "Manage Palette"
2. Version 3.0.0-beta.3 gets installed
3. Configure S7 node
4. Don't connect the PLC -> Status of the node is "offline"
5. Connect PLC -> node-red crashes

### Expected behavior
It is possible to establish a connection to a PLC with the version which get automatically downloaded

### Screenshots
If applicable, add screenshots to help explain your problem.

### Logs
```
- TypeError: stream.off is not a function
at ISOOnTCPClient.isoOnTcp.createConnection (/usr/lib/node_modules/node-red-                                                                                                             contrib-s7/node_modules/@st-one-io/nodes7/src/s7endpoint.js:207:24)
at Object.onceWrapper (events.js:313:30)
at emitNone (events.js:106:13)
at ISOOnTCPClient.emit (events.js:208:7)
at process.nextTick (/usr/lib/node_modules/node-red-contrib-s7/node_modules/                                                                                                             iso-on-tcp/src/client.js:193:45)
at _combinedTickCallback (internal/process/next_tick.js:132:7)
```

### Flow
Please attach your flow either as a file or properly formatted as JSON

[flows_iot2000.zip](https://github.com/st-one-io/node-red-contrib-s7/files/5517088/flows_iot2000.zip)

### Environment
 - **Device**: IOT2040 with Yocto 2.6
 - **Version** of this node: 3.0.0-beta.3
 - **Node-RED version**: V1.2.3
 - **PLC Part Number**: tested a S7-1200 and a S7-1500
 - **OS**: Windows 10
 - **Browser**: Chrome

### Additional context
With the IOT2050 (Debian 10 based) it doesn't result in an uncaught exception and everything works fine

gfcittolin, 2020-11-30:
Thanks for the report. Unfortunately, for 3.0.0 onwards only versions of nodejs 10 or greater will be supported, due to big changes in the underlying codebase. I'll put a notice on the readme, so that this requirement is documented and explicit.

closed 2020-11-30 by bergmanu

RicardoCotrina, 2021-07-04:
Gracias por el aporte estimado, pero en donde te ubicas dentro del SO YOCTO para que ejecutes ese comando ? Dentro de 
".node-red" ? O en donde, si pudiera aclarar ello por favor, gracias de antemano. Saludos