st-one-io/node-red-contrib-s7 #84: I have translated your library into Italian, can you add the translation to the original repository?

ivan0x16, 2020-12-20:
I have translated your library into Italian, can you add the translation to the original repository?

thank you very much for your work
the library works great!

Tested with Tia portal V14 , node-red 1.0.3

repository: node-red-contrib-s7-Italian-translation
directory: node-red-contrib-s7-Italian-translation/red/locales/it-IT/s7.json

Best regards

__Originally posted by @ivan0x16 in https://github.com/st-one-io/node-red-contrib-s7/issues/80__

fernandoamorim1703, 2022-06-10:
Hello, we saw that you created a PR, but it closed in March. If you can create another PR so we can analyze and add the Italian translation in node

fernandoamorim1703, 2022-06-24:
We will be closing this issue, if there is a need to revisit the subject, please open a new one. Thank you very much.

closed 2022-06-24 by fernandoamorim1703