st-one-io/node-red-contrib-s7 #89: stream.off is not a function

Venomz101, 2021-03-05:
### Describe the bug
When creating a connection and reading tags from my PLC with NodeRED on Siemens IOT2040, it crashes my NodeRED as soon as the flow is deployed and connection is being attempted.

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior
A clear and concise description of what you expected to happen.

### Screenshots
[image]
[image]

### Logs
If applicable, attach the Node-RED logs with the "Verbose" option on the configuration node set to "On"

### Flow
[{"id":"2307bc4c.859144","type":"tab","label":"Flow 1","disabled":false,"info":""},{"id":"942afe1f.57eb1","type":"s7 endpoint","z":"","transport":"iso-on-tcp","address":"192.168.200.10","port":"102","rack":"0","slot":"1","localtsaphi":"01","localtsaplo":"00","remotetsaphi":"01","remotetsaplo":"00","connmode":"rack-slot","adapter":"","busaddr":"2","cycletime":"500","timeout":"2000","name":"","vartable":[{"addr":"DB100,REAL0","name":"TestReal 1"},{"addr":"DB100,REAL4","name":"TestReal 2"},{"addr":"DB100,REAL8","name":"TestReal 3"}]},{"id":"2fa54dac.18cff2","type":"s7 in","z":"2307bc4c.859144","endpoint":"942afe1f.57eb1","mode":"all","variable":"","diff":true,"name":"S7 Read","x":370,"y":280,"wires":[["7c218671.836568"]]},{"id":"7c218671.836568","type":"file","z":"2307bc4c.859144","name":"StoreToFile","filename":"/home/files/S7Output","appendNewline":true,"createDir":true,"overwriteFile":"false","encoding":"none","x":600,"y":280,"wires":[[]]}]

### Environment
 - **Version** of this node: 3.0.0-beta.3
 - **Node-RED version**: 0.20.7 / node.js 8.16.0
 - **PLC Part Number**: [e.g. 6E57-7512 ]
 - **OS**: Linux 4.4.185-cip35 ia32 LE
 - **Browser**: chrome

### Additional context
Add any other context about the problem here.

Venomz101, 2021-03-05:
Found just now the requirements for the latest package!
[image]
Seems like this is probably what's causing the issue.

fernandoamorim1703, 2023-07-28:
Hello @Venomz101 , 
as the comment above suggests, try updating your Node-RED and Node.js versions.

This node was designed for Node-RED version 1.1.3 and Node.js version 14.

We will be closing this issue, if there is a need to revisit the subject, please open a new one. Thank you very much.

closed 2023-07-28 by fernandoamorim1703