st-one-io/node-red-contrib-s7 #91: ECONNRESET when PLC connection drops

matteoangrisano, 2021-04-07:
Hello,
I have a problem with Node-RED:
continuously communicates with PLCs using node-red-contrib-s7, it worked well for a long time, once migrated Node.JS and Node-RED (without changes to the flows) from a machine with Windows 7 to a machine with Windows Server 2012 r2, I started getting the following error after a day or two:

16 Mar 10:33:01 - Error: read ECONNRESET
at TCP.onStreamRead (node: internal / stream_base_commons: 211: 20)

this error stops Node-RED

Node.js version: 15.11.0
npm version: 7.6.1
Node-RED version: 1.2.9

I also tried with Node.js V14 but the error reoccurred.

I noticed that the problem occurs when the connection with a PLC is missing for more than a few minutes.

Last part of the logs:

7 Apr 11:30:39 - [error] [s7 endpoint:PLC - Pozzi Vasca Vecchia] Error: Timeout connecting to the transport
7 Apr 11:30:40 - [error] [s7 endpoint:PLC - Pozzi Vasca Nuova] Error: Timeout connecting to the transport
7 Apr 11:30:54 - [error] [s7 endpoint:PLC - Pozzi Vasca Vecchia] Error: Timeout connecting to the transport
7 Apr 11:30:55 - [error] [s7 endpoint:PLC - Pozzi Vasca Nuova] Error: Timeout connecting to the transport
7 Apr 11:31:10 - [error] [s7 endpoint:PLC - Pozzi Vasca Nuova] Error: Timeout connecting to the transport
7 Apr 11:31:13 - [red] Uncaught Exception:
7 Apr 11:31:13 - Error: read ECONNRESET
at TCP.onStreamRead (internal/stream_base_commons.js:209:20)

gfcittolin, 2021-04-14:
Have you updated the node-red-contrib-s7's version as well in the process, or just the environment has been updated? If yes, which was the previous and what is the current version?

matteoangrisano, 2021-04-15:
> Have you updated the node-red-contrib-s7's version as well in the process, or just the environment has been updated? If yes, which was the previous and what is the current version?

I probably also updated the version of node-red-contrib-s7, unfortunately I don't remember which one was the old one.

capIdxItaly, 2021-05-05:
Same probelm with library "plcindustry v2.1.1"
Node.js version: 14.15.4
npm version: 6.14.10
Node-RED version: 1.2.7

and with 
Node.js version: 14.16.0
npm version: 6.14.11
Node-RED version: 1.2.9

The problems show itself (for me) only on startup if a node is saved in memory. If i cancel the node from the flow file and after that I'll import the same node deletd earlier it stays on like it's supposed to be.

matteoangrisano, 2021-05-12:
@gfcittolin any update for this issue?

coste082, 2021-11-17:
@gfcittolin I am having this issue as well. s7-in node reports "ECONNRESET" after PLC drops connectivity and does not resume connectivity when PLC is reconnected.  As capIdxItaly suggested, if i remove flow from file and reimport and deploy, the node works as designed. Do you have any updates?

coste082, 2021-11-18:
@gfcittolin I found this update for the nodeS7 this project was built on. Could this update be implemented here?
https://github.com/plcpeople/nodeS7/commit/4a0c964a8eeccea145190bda2c277e4c0a1c6317

matteoangrisano, 2021-12-07:
> @gfcittolin I found this update for the nodeS7 this project was built on. Could this update be implemented here? [plcpeople/nodeS7@4a0c964](https://github.com/plcpeople/nodeS7/commit/4a0c964a8eeccea145190bda2c277e4c0a1c6317)

@gfcittolin is this possible?

gfcittolin, 2021-12-22:
The fix of plcpeople's nodeS7 is not applicable here, since we don't use it anymore from version 3.0.0 onwards.

The problem with unhandled `ECONNRESET` on Node-RED is that it can come from any node that handles TCP connection, not necessarily this one. Had a problem with a customer last week regarding `ECONNRESET`s killing Node-RED, and the culprit was a modbus node, even though the connection to the PLC was unstable too. 

Although I wouldn't rule it out that it comes from this node, I've checked the error handling paths and there's no obvious and not-so-obvious missing error handling. I couldn't also simulate the issue, although specific network conditions and timings could be critical for reproducing the issue.

Could you guys please confirm if the issue happens in a flow with a single `s7 in` node and nothing else?

digidax, 2024-09-03:
Same problem here:
`Sep  2 16:07:35 173 Node-RED[2835544]: 2 Sep 16:07:35 - [error] [s7 endpoint:Dampfsteuerung] Error: Timeout connecting to the transport
Sep  2 16:07:38 173 Node-RED[2835544]: 2 Sep 16:07:38 - [red] Uncaught Exception:
Sep  2 16:07:38 173 Node-RED[2835544]: 2 Sep 16:07:38 - [error] Error: read ECONNRESET
Sep  2 16:07:38 173 Node-RED[2835544]:    at TCP.onStreamRead (node:internal/stream_base_commons:217:20)
Sep  2 16:07:38 173 systemd[1]: nodered.service: Main process exited, code=exited, status=1/FAILURE`

Problem focus is in S7 node:
[https://discourse.nodered.org/t/node-red-unexpected-restart/90668](url)
**In short, if the author of a node does not handle exceptions it can crash the node process.**

Please handle this exceptions in this node: 
- Timeout connecting to the transport
- connect EHOSTUNREACH

Thanks a lot.