st-one-io/node-red-contrib-s7 #93: Problems Reading MD memory Variables

motosierra1141, 2021-06-17:
### Describe the bug
I have been trying to connect my PLC to node red, i can read variables like boolean and int, but when i want to read a MD variable a memory that is a Real variable it show a number like 1077095984 a huge number, when i connect to the PLC to see the real value of variables i can see that the value it is very different to the number i see in node red
### To Reproduce
1.- try to read a Variable type Real that it is a MD a memory i use a MDIXXX in node red
### Expected behavior
I try to see with debug the value that i see in the PLC monitoring the variables in Tia Portal

### Screenshots
[image]
in  tia portal this variable it is between 2.5 and 4 

if you click on the variable in node red it change into a date format, very weird
[image]

### Logs
If applicable, attach the Node-RED logs with the "Verbose" option on the configuration node set to "On"

### Flow
[{"id":"a4e62aa9.84b5c8","type":"tab","label":"Flow 2","disabled":false,"info":""},{"id":"bce8a775.a8aad8","type":"debug","z":"a4e62aa9.84b5c8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":400,"y":140,"wires":[]},{"id":"ac63e5ae.fb3d88","type":"s7 in","z":"a4e62aa9.84b5c8","endpoint":"42ad9527.92dc0c","mode":"single","variable":"PRES_ABS_RIN","diff":false,"name":"","x":210,"y":160,"wires":[["bce8a775.a8aad8"]]},{"id":"42ad9527.92dc0c","type":"s7 endpoint","transport":"iso-on-tcp","address":"192.168.0.4","port":"102","rack":"0","slot":"1","localtsaphi":"01","localtsaplo":"00","remotetsaphi":"01","remotetsaplo":"00","connmode":"rack-slot","adapter":"","busaddr":"2","cycletime":"2000","timeout":"1000","name":"LLENADORA_2","vartable":[{"addr":"MDI144","name":"VEL_LLN"},{"addr":"MDI152","name":"VEL_NORM"},{"addr":"MDI232","name":"PRES_RIN"},{"addr":"MDI266","name":"PRES_ABS_RIN"},{"addr":"MDI140","name":"VELOCIDAD_TRANSPORTE"},{"addr":"MD266","name":"presionprueba"},{"addr":"","name":""}]}]

### Environment
 - **Node-RED version**: 1.2.9
 - **PLC Part Number**: Siemens S-1200 
 - **OS**: UBUNTU
 - **Browser**: Chrome

### Additional context

gfcittolin, 2021-06-17:
Hi there, try using `MRxxx` for reading `REAL`-typed memories.

[image]

motosierra1141, 2021-06-17:
Thanks You so much!

closed 2021-06-30 by gfcittolin