st-one-io/node-red-contrib-s7 #96: Reading endpoint configuration from a text file

TommyBerz, 2021-07-26:
Hello I am trying to set the endpoint configuration throught the reading of a text file (stored in my PC) that contains the IP and the port useful for the endpoint configuration. I did the file reading and the recovery of the IP/port but I can't set them as endpoint connection parameters. It would be useful if you can read parameters directly from the payload, with that we could change endpoint configuration dynamically.

gfcittolin, 2021-07-28:
Right now this is a bit out of sight for this node, because of the structural changes this would imply to the code and some edge cases that we'd need to sort out. Since, if you need dynamic endpoints you'll probably need dynamic variables too, doesn't it make sense to use the nodes7 library directly?

TommyBerz, 2021-07-28:
Which library in particular are you thinking about?

gfcittolin, 2021-07-29:
Either [@st-one-io/nodes7](https://github.com/st-one-io/nodes7), the library that powers this Node-RED node, or [plcpeople/nodeS7](https://github.com/plcpeople/nodeS7), the original one that was the inspiration for the former

closed 2021-07-30 by TommyBerz